HclX/WyzeHacks #5: SD card shows as "0.05 GB" on some cameras

ctjameson, 2019-11-15:
Install works, folder appears on NAS, however the SD card shows a capacity of 0.05 GB instead of the 14TB that the others show. Still shows the same after rebooting a dozen times.

Running Wyze Cam v2 on latest commit on newest Wyze firmware (4.9.5.35)

ctjameson, 2019-11-20:
Looks to be an issue with the newest firmware (4.9.5.36) as all cameras are now showing the same thing and none are recording. I guess that just shows I should have stayed on a lower FW. :/

pumasight, 2019-11-20:
Now that they have a streaming  service....

ctjameson, 2019-11-20:
Yup. Guess I need to find the older firmware somewhere and set them to
never update. :/

ctjameson, 2019-11-20:
Reverted to 4.9.4.169 and everything is great again! Now to pull down all my cameras yet again. -_-

closed 2019-11-20 by ctjameson

stephenjamieson, 2019-12-05:
I had this issue with mounts above 2tb.  When I put a quota of 2tb on the ZFS dataset I was using for the nfs share it worked fine.  I tried 3tb and it broke, not sure what the limit is but 2tb is fine for me.  Note that it also broke `df`.

ctjameson, 2019-12-05:
It’s actually fully fixed now in the newest revision so we’re all good.

stephenjamieson, 2019-12-05:
Interesting, I definitely can't mount anything above 2tb on the latest but that's probably a limitation of the embedded version of linux it's running.

ctjameson, 2019-12-05:
Weird. Mine is mounted to a 14TB NFS share on my Synology without any
issue. Newest firmware, newest hacks version.