HclX/WyzeHacks #6: NFS Connection Refused error

kdvlr, 2019-12-01:
I was getting connection refused error while trying to mount my nfs server when I tried to manually mount my NFS drive. I thought it was an error on my linux server, so I tried my old NAS and that still showed a problem. Other clients were connecting, but WyzeCamPan was not connecting. 

I forced it to connect via UDP using ```-o port=2049,nolock,proto=tcp``` and this mounted the drive immediately.

closed 2019-12-01 by kdvlr

FuzzyMistborn, 2020-01-25:
Can you explain a bit more how you accomplished this?  I've gotten telnet working but not NFS mount to my Synology.

kdvlr, 2020-01-26:
What’s the error you get when you try to mount manually ?

FuzzyMistborn, 2020-01-26:
```
          inet addr:192.168.10.49  Bcast:192.168.10.255  Mask:255.255.255.0
 not mounted, try mounting to /mnt...
 on /mnt failed: Permission deniedume1/Wyze
 /mnt] failed, will retry....168.10.20:/volume1/Wyze

```

The mount line in config.inc is:
`export NFS_ROOT='192.168.10.20:/volume1/Wyze'`

I've confirmed that I can ping/telnet to the Synology on ports 111 and 2049.

kdvlr, 2020-01-26:
Seems like a permission issue. Were you able to mount the NFS share on any other machine ?

FuzzyMistborn, 2020-01-26:
Yep I can mount it from another device no issues.

FuzzyMistborn, 2020-01-27:
Ok so i managed to get it to mount via telnet with

` mount -o port=2049,nolock,proto=tcp 192.168.10.20:/volume1/cameras /tmp/test`

Now how do I get it to do that automatically with the script?  I'm still getting permission denied when i run ./mount_nfs.sh