HclX/WyzeHacks #7: no nfs mounted, but root is working

tomatensaus, 2019-12-07:
I followed your instructions

cd /system/bin/

cat mount.sh 
#mount -o nolock,rw 192.168.31.218:/home/xuxuequan/Ingenicwork/sharenfs /mnt

Looks like my NFS from the config file was ignored

tomatensaus, 2019-12-07:
I tried to mount my nfs without success. Found the issue with the path

telnet 
cd /system/wyze_hack
fixed the path in config.inc
source config.inc
./mount_nfs.sh

Up and running!

HclX, 2019-12-19:
yes, you need to have your config.inc. When you install the hack, you can put your config.inc in the root of the sd card, it will get copied onto the camera during installation.  Later installation will inherit the previous configuration file.

closed 2019-12-19 by HclX

chaoticchoas, 2019-12-21:
So I am having a similar issue. I configured the config.inc and have it in the root of the sd card but it seems like the hack is not using it. I attempted to do what @tomatensaus did but I continue to get a permission denied error when attempting to run ./mount_nfs.sh.
`Mounting directory /mnt/WyzeCams/macaddr as SD card
Creating data directory [/mnt/WyzeCams/macaddr]
mkdir: can't create directory '/mnt/WyzeCams/': Permission denied
[mkdir -p /mnt/WyzeCams/macaddr] failed, will retry...
`
This is on a wyze cam v2 running on firmware 4.9.5.36 as well as V4.9.4.108

FuzzyMistborn, 2020-01-25:
I'm having a similar problem trying to mount my synology NFS share.  I keep getting permission denied.  Any ideas?

tomatensaus, 2020-01-26:
@FuzzyMistborn  try with telnet... then mount by hand to get the path right.
In my case I had to add /export/ to my path eg mount -t nfs /export/WyzeCam /tmp/test1

FuzzyMistborn, 2020-01-27:
I got mounting by hand to work with:

`mount -o port=2049,nolock,proto=tcp 192.168.10.20:/volume1/cameras /tmp/test`

I tried modifying the mount_nfs.sh script to add those options/arguments but still getting permission denied.

tomatensaus, 2020-01-27:
seems like the script in /system/bin/mount.sh is not used and the one in wyze hacks is used

edit the /system/wyze_hack/mount_nfs.sh

edit the mount command
NFS_MOUNT="/bin/mount -o nolock,rw" to match your requirements

FuzzyMistborn, 2020-01-27:
Hmmm, that's the one i edited (in /system/wyze_hacks).  I'll try again.