HclX/WyzeHacks #8: Restart Wyze if NFS Mount is stale or not reachable ?

kdvlr, 2019-12-15:
What's the best way to restart the camera if the NFS is not working?

For now,I am using ``` if ! timeout -t 10 df -h | grep -q /media/mmcblk0p1 ; then reboot -f -d 10;fi``` to reboot if NFS connection is stale or if my NFS server had restarted. However, if NFS is unreachable for a while then there is a reboot loop unless there is a local file to keep track of restarts. 

Is there a better way to test if an NFS share is available again after a disruption and only then restart the camera. Normally you could do this with showmount, but that does not exist on the wyzecam.

HclX, 2019-12-19:
i believe the current behavior in my script is to keep trying till it the NFS share mounts. So if you add another loop at the end of mount_nfs.sh with your command, you will be able to detect the NFS stale connection and reboot the device.

If your NFS server is down for a while, at next reboot, it will loop infinitely trying to mount the share. Thus you will not be in a reboot loop.

kdvlr, 2019-12-20:
Thanks. I have added it to log_sync.sh because mount_nfs doesn't seem to be continuously running on my install. You're right on the reboot loop. 

However, if I check for the stale message, it doesn't seem to show up until the NFS connects again. 

The following works for me. It doesn't restart unless the NFS server is back up again. 
```
 if ( timeout -t 60 df -h 2>&1| grep -q 'Stale NFS');
    then
        reboot -f -d 10
    fi
```

closed 2019-12-20 by kdvlr