HclX/WyzeHacks #10: Can't telnet after installing

netjeff, 2019-12-31:
I think I followed the instructions correctly to update my WyzeCam v2, but putty telnet says "connection refused".   Probably user error on my part, but I can't spot my problem.

More details ...

1. Started with WyzeCam v2 running stock 4.9.5.36 firmware.

2. Copied the appropriate files from https://github.com/HclX/WyzeHacks/tree/master/release into the root folder of an empty 2GB sd card with FAT32 format.  Including rename+edit of ``config.inc.TEMPLATE``, the root of the sd card looks like this:
```
g:\>dir
 Volume in drive G is SD2GB
 Volume Serial Number is 5CBA-50FE

 Directory of g:\

12/31/2019  12:42 PM           112,640 FIRMWARE_660R.bin
12/31/2019  12:42 PM                65 version.ini
12/31/2019  12:34 PM             1,347 config.inc
               3 File(s)        114,052 bytes
               0 Dir(s)   1,969,696,768 bytes free
```

3. With WyzeCam unplugged from power, inserted SD card.

4. Plugged in WyzeCam, waited for solid blue light

5. After solid blue light, camera working fine in app.

6. Used app's "device info" to get IP address of cam.

7. Used putty to try and telnet to that IP address.  Result is "connection refused"

==============

Issue is probably user error on my part, but I can't spot what I'm doing wrong.

tomatensaus, 2020-01-06:
If the upgrade ran the version.ini will be renamed to version.ini.old
Looks like your upgrade did not run. I also think the camera says via speaker "setup complete"

netjeff, 2020-01-09:
I opened this issue while trying on my first Wzye Cam v2, that I've had for about a year.

I just got another Wyze Cam v2, so I tried the same on that one.  It worked perfectly.  I was able to telnet into the second cam.  Also I saw that on the sd card the version.ini file was now named version.ini.old.

So its something about my first Wyze Cam v2.   FYI, a few weeks ago I tried the RTSP firmware from Wyze, but it wasn't working well for me.  So I flashed back to the stock 4.9.5.36.  And then tried this and it didn't work, so I wrote the issue.

Just in case, on my first cam I did factory reset, again flashed the stock 4.9.5.36, and factory reset one more time.  Than tried the sd card with this.  Waited for the solid blue.  But the app could not connect.  While giving the app some time, noticed that it seemed like the cam rebooted itself, because I saw the sequence of solid orange, flashing blue/yellow, then back to solid blue.  But still app would not connect.

So I unplugged, removed sd, plugged in.  After solid blue (with no sd card) app connected fine again to first cam.  But no telnet.  Also checked the card and the version.ini was still named version.ini.

So it seems like something with my first cam is having problems with this?  Any ideas for further troubleshooting?  Good news is that my second cam is working with this, but would love to get my first cam working, too.

tomatensaus, 2020-01-23:
Found this comment about renaming to demo.bin .... then renaming back, might work for you

https://www.reddit.com/r/wyzecam/comments/cpzlwf/directly_record_to_your_nfs_share_no_need_to_wait/exdggp0?utm_source=share&utm_medium=web2x

HclX, 2020-01-27:
with the latest release, I added some voice prompts indicating the starting and ending of the installation . If you are using latest release and didn't hear anything, the installation wasn't even started.

If you heard the starting sound but not the finishing sound, you can try creating a directory named "debug" at the root of the SD card and try it again. It will copy a couple important files into that directory for debugging purpose, including the logging output from installation script. It may give you some hints on why it's failing.

HclX, 2020-01-27:
btw, do not share the content in the "debug" directory with anyone, it has your per device credential which allows people to "clone" a wyze camera.

thompatry, 2020-03-22:
Here are the steps for me to get it work on firmware version 4.9.5.98.

1. Flash the Wyze firmware and set up camera in Wyze App. Make sure everything is working here.
2. Copy the 3 files from the readme steps.
3. Modify your config.inc as suited
4. With the camera on, insert the micro sdcard and wait. You will hear some sounds including "Setup Completed." The camera should reboot. Only takes about 30-60 seconds to complete.
5. Now try telneting. It should work.

closed 2020-06-12 by HclX