HclX/WyzeHacks #14: /mnt is read only, getting permission denied when trying to mount

FuzzyMistborn, 2020-01-30:
I can mount my NFS manually with `mount -o port=2049,nolock,proto=tcp 192.168.10.20:/volume1/cameras /tmp/test` via telnet in the camera.  However, whenever I try to modify mount_nfs.sh to add the required options, I always get permission denied when I try to run the script.  Not sure if i'm just not editing the script correctly.

I can't create a directory in /mnt as it's read-only, wondering if that's the issue (though I can mount by hand)?  Really want to take advantage of this hack, any help would be appreciated.

HclX, 2020-02-03:
/mnt itself is a directory of sqashfs, and its readonly by default. However, if you mount NFS share, it should be writable -- if you have your NFS permission configured correctly. I'd suggest you figure out the exact options you use on another linux computer first, and then apply the same to the wyze hack script.

FuzzyMistborn, 2020-02-03:
I do, I've tested the mount on another linux box and I can read/write files just fine on the same mount with just a standard `mount 192.168.10.20:/volume1/cameras /path/to/dir` .  The Wyze seems to require the extra options, but even modified with the script it's not working.  Hence where i'm getting stuck.

FuzzyMistborn, 2020-03-08:
Figured out the problem.  Weird ^M character in the config.inc file.  Removing them via vi and reloading source config.inc and running ./mount_nfs.sh worked.

[image]

closed 2020-03-08 by FuzzyMistborn