HclX/WyzeHacks #15: wyze hack doesn't persist through the update of 4.10.5.98

HclX, 2020-02-03:
I was checking the latest update and noticed that it's using a rather complicated approach which causes wyzehack no longer persist through the update, will looking for solutions.

At this moment, I think the solution would be re-install the hack: for camv2 the telnet interface should still open after update so you can simply telnet into it and install the hack. However, for cam pan, even though the telnet is still active, the credential is unknown so you will have to use the SD card method.

HclX, 2020-02-05:
working on this, stay tuned... meanwhile, reinstall wyze hack after upgrade, you can do that with telnet remotely if it's not a PAN.

mpatton125, 2020-03-23:
I have this problem after upgrading to the latest (non-pan) firmware 4.9.5.111.
Telnet is still fine, but no NFS.  

Can you please advise how to re-install the hack over telnet?

thompatry, 2020-03-25:
Did you try running install.sh from the the NFS share again with the config file?

mpatton125, 2020-03-25:
Not sure what would be the point - the NFS share isn't mounted on the cam after the update?

thompatry, 2020-03-25:
Crap. I wasn’t thinking. Let me go back to the drawing board. The easiest
way is to factory reset and reinstall the hack.

mpatton125, 2020-03-25:
All good.  I just re-installed the hack the normal way with the sdcard- didn't need to factory reset.

HclX, 2020-03-27:
yes, "sdcard install" after a firmware update is the best way to get it working again. You may be able to manually mount nfs share and then run install.sh from the share directly, but that path isn't thoroughly tested so i can't promise it will work.

thompatry, 2020-03-28:
Another way I have got it to work is to stand up a webserver with the install.sh, config.inc, the FIRMWARE file, and version.ini. Then wget to the /tmp directory and run chmod a+x install.sh. Then run ./install.sh.

HclX, 2020-03-29:
ok, with latest code now it should survive through firmware update, at least from 4.10.5.98 to 4.10.5.111. I'm closing this one now.

closed 2020-03-29 by HclX

thompatry, 2020-04-30:
Just a FYI, Wyze is preparing to release a new firmware. It is in beta right now. It breaks telnet and NFS. The firmware is listed below.  Do not upgrade. I assume the wyze overwrote the .Server_config which enables telnet and did some changes on how the SD Card is being mounted.

I have not tested reinstalling the hack yet. I just tested the firmware tonight. @HclX Let me know if you want me to grab anything from camera after I install the hack.

Here are the versions and change log. (Kind of vague)
v2: 4.9.6.59
Pan: 4.10.6.59

-     Fixed several issues causing camera rebooting
-     Fixed a Wyze Cam Pan two-way audio issue
-     Fixed an audio noise issue in Wyze Cam v2 live stream
-     Fixed a bug causing incorrect camera settings after firmware update
-     Fixed a bug impacting Wyze Sense Bridge functionality

HclX, 2020-05-06:
thanks for the heads up. I have access to the beta firmware, but my home box got a power supply failure, i'm still waiting for the new power supply to start working on it....

thompatry, 2020-05-06:
I haven't really dug into it yet but I downgraded from the app today back to the public firmware 4.9.5.111 and telnet/nfs mounting came back to life without me having to reinstall the hack. Not sure if it is a beta firmware thing or if something also changed where the telnet daemon or the nfs mount module wasn't compiled.

thompatry, 2020-05-06:
To expand again, Wyze gave a little bit of insight of why NFS and Telnet did not work on the new beta. Good news, it won't be released for a little while since they are so many bugs in it. But it comes down to that they did a complete firmware rewrite to make the code more efficient. So in the sense, there might have to be some reworking done before anyone upgrades.