HclX/WyzeHacks #16: Flashed Multiple times but still no NFS

StanESmith, 2020-02-08:
I've flashed the hack multiple times and each time with a different response. Seems as though the hack runs into errors and doesn't have an error catch. Sometimes I flash the hack on 4.9.5.36 and get telnetd and NFS but not mounted in the SDCARD location, other times I get only telnetd and other times the camera reboots without applying anything. 

Not really sure what's needed or how to execute the hack when I get the camera to boot again with only telnetd. I've tried to look through the source code here trying to apply the hack manually at times when I only get telnet but I have not had success.

HclX, 2020-02-28:
The hack is designed to add NFS functionality with minimum impact to the firmware's original feature. Due to that, the hack is running in a not very straightforward manner so it's hard to tell what's gone wrong without looking into each specific case.

I'd go through the following steps debugging what's wrong:
1. After hack installation, make sure you reboot the camera WITHOUT physical SD card inserted. I've found multiple instances that the presence of a physical SD card is causing problems.

2. Telnet into the device, and check if the hack is correctly installed, which can be done by "ls -l /system/init/app_init.sh" -- if the file is showing up as a symbolic link, the hack is correctly installed, otherwise something went wrong during installation.

3. If the hack is correctly installed, all the console logs are now redirected to /tmp/boot.log. Copy that file out and look for suspicious error messages. The log is quite noisy so you need to check the wyze hack script and search for the error message.

4. If the logs says failed to mount NFS share, try manually mount it using the same command.

I try to make the hack as easy to use as possible, however, when things go wrong, you will need plenty of linux knowledge to debug it.

HclX, 2020-03-27:
make sure your config.inc file is with linux line ending, which means if you created the file with notepad on windows it will not work. To verify that, run "vi config.inc" from linux and if you see ^M at the end of each line, your line endings are windows line ending and it will not work.

At this moment, use notepad2 or other editor software to change the configuration file to linux line ending. I'm going to fix that in next release.

HclX, 2020-03-29:
now with latest code config file with windows line ending should work.

closed 2020-06-12 by HclX