HclX/WyzeHacks #17: NFS Filesystem Type - FAT32 or NTFS

thompatry, 2020-03-22:
Are you using FAT32 or NTFS for your NFS share. I can get my share to mount but no record folder is created. I run 'df -h' I get the error 'Value too large for defined data type'. This leads me to think  large file support is not part of the Wyze firmware. Any thoughts?

I am using NFS4 with a 8 TB drive.

thompatry, 2020-03-22:
So my assumption was wrong. It has nothing to do with NTFS vs FAT32. In theory it should not matter. I reran my chown/chmod commands it appears it fixed it. Not sure why.

thompatry, 2020-03-22:
From my testing of different filesystems, it must be a ext* or fat32. You cannot use NTFS for this. Can someone confirm?

HclX, 2020-03-27:
i didn't try NTFS. As my understanding, it's a network file system, so the actual disk format shouldn't matter. "df -h" may fail on a too large volume but that shouldn't affect the actual use, except that you may not see the correct size from wyze app.

thompatry, 2020-03-27:
"df -h" did not work right as stated and within the Wyze app, it didn't
show a size. It also did not work as expected with a large NFS share.
~~
Thomas Patry
thompatry@gmail.com
(785)-643-0997

thompatry, 2020-03-28:
I will close this since this isn't huge issue.

closed 2020-03-28 by thompatry