HclX/WyzeHacks #18: Camera not recording to NFS Share

barthohomeautomation, 2020-04-01:
So I went through the complete process, and all seems to be good. I am able to connect to the camera, and confirmed the NFS share is mounted. However, video is not being recorded to the NFS share. I did confirm the "Continues Recording" option is selected. I can see in the folder "Alarm" images are being recorded so it seems it is not a permissions issue. But I do not see video anywhere. Also, then I go to video playback it shows no video is available. I'm assuming this is user error in some way shape or form but I do not know where to look to resolve. Can someone shed some light on what the issue may be? Thanks in advance.

barthohomeautomation, 2020-04-02:
I have resolved the issue. This was caused by my VPN.

closed 2020-04-02 by barthohomeautomation

agrieco, 2020-04-06:
I’m having exactly the same issue w/ no VPN.

I see alarm and time lapse folders created, jpgs go into alarm, but no video files anywhere even though the camera is on continuous recording.  Any ideas?