HclX/WyzeHacks #19: No Video on NFS

agrieco, 2020-04-07:
Having the same issue as documented here: https://github.com/HclX/WyzeHacks/issues/18 but mine hasn't gone away. 

I see a time_lapse, alarm directory and a few text files in the target NFS share- so I know its writing files fine.  within the alarm directory, I see new directories created every day, with a couple of jpgs in them.

Any idea what gives?
thanks

thompatry, 2020-04-07:
Just a few questions, what size is your NFS share? What is the partition type? (NFTS, FAT32, etc)

agrieco, 2020-04-08:
ZFS is the underlying FS.   Size is 4T

thompatry, 2020-04-08:
Do you have a small flash drive or external that is 1TB or less? If so, create a ZFS partition on it and make a NFS share and report back. 

I had a weird issues just like you but I was using an 8TB Ext4 partition and it would create the folders and such but never record anything. I ended up partitioning off 1TB partition on the drive and set a NFS share for that and it works. I think the camera kernel cannot handle very large drives.

agrieco, 2020-04-08:
That must be it.  I don't have an easy way to create a small ZFS partition but created a share on a different host with ext4 partition and it worked like a champ.

This is great- thanks for the work.  I wonder if there is any chance of getting to the bottom of the blocking using a larger partition?

thompatry, 2020-04-10:
It could be a kernel limitation. Someone would have to dig in and look.

alexschneider, 2020-04-25:
In case anyone else is using FreeBSD / ZFS like me, I found a workaround using the instructions here: https://www.freebsd.org/doc/handbook/disks-virtual.html

I think that you lose a lot of the benefits of ZFS by doing this, and I have no idea how resilient the recordings are at this point (so you might want to have a cronjob copy them somewhere else on your filesystem periodically)

```
# this command will create a file for backing your filesystem that's 512gb. If you want larger or smaller just adjust the bs/count numbers
$ dd if=/dev/zero of=CameraFS bs=1G count=512 # this took about 4 minutes for me
# If you get an error you can change `-u 1` to `-u 2` or `-u 3` or so on. Make sure to change `md1` in future commands to the number you picked here
$ mdconfig -f CameraFS -u 1
$ bsdlabel -w md1 auto
$ newfs -U md1a
# Replace `Cameras` with whatever folder you've set up for NFS. Make sure to move all content in the folder prior to running this command (the folder must be empty).
$ mount /dev/md1a Cameras
# You will likely have to fix permissions after doing this
$ chmod -R alex Cameras/
```

After this is finished, you'll need to restart all your cameras so they start writing to the right location.

stephenjamieson, 2020-06-03:
:wave: , you can set a quota which works as well.  

```
zfs set quota=512G path
```

closed 2020-06-12 by HclX