HclX/WyzeHacks #20: Install Trouble

dlashua, 2020-04-08:
I have prepared the SD card as indicated in the README. Format FAT32. Extract the release to the root of the card, modify config.inc.

The instructions are slightly less clear on what to do next. I've tried the following.

1) With the camera powered on, just insert the card and wait. Nothing happens.

2) With the camera off, insert the card. Then power the camera. Firmware does not get installed.

3) With the camera off, insert the card. Then hold "RESET" while powering the camera until the LED changes colors. After a very long time, this resulted in the "Installation complete. Remove SD Card" message.

If step 3 is the correct procedure, the README should be updated to reflect that. 

Continuing from 3, the instructions say to remove the card. I removed it, but was unable to connect via telnet. So then I restarted the camera, and I'm still unable to connect.

dlashua, 2020-04-08:
I tried several more things and didn't keep track. I was running a watch on my NFS directory waiting for the camera to create a directory. But, somewhere in all of that, telnet started working.

NFS doesn't appear to be mounting. However, with telnet I'm able to try a few things.

I did this:
```
cd /tmp/run/wyze_hack/
export NFS_ROOT=`192.168.0.7:/wyze/`
./mount_nfs.sh
```
I get this message, which takes 5 seconds or so to produce:
```
          inet addr:192.168.0.91  Bcast:192.168.0.255  Mask:255.255.255.0
192.168.0.7:/wyze/ not mounted, try mounting to /mnt...
mount: mounting 192.168.0.7:/wyze/ on /mnt failed: Permission denied
[/bin/mount -o nolock,rw 192.168.0.7:/wyze/ /mnt] failed, will retry...
```
Which repeats over and over.

I am able to set THIS_DIR and run install.sh in the same directory. However, it doesn't seem to fix the NFS mounting issue.

dlashua, 2020-04-08:
I figured this out. It seems that even thought this will connect to an nfs4 server, it doesn't support fsid=0. In other words, you have to provide the full path to the nfs share. For me this meant:

```
export NFS_ROOT='192.168.0.7:/srv/nfs/wyze/`
```

HclX, 2020-06-12:
Simply inserting the SD card is enough, you shouldn't need to reset or doing anything unless the camera has a previous version of wyzehacks installed.

HclX, 2020-06-12:
also, you want to update the config file with your NFS settings. On device, the latest version config file is located at /params/wyzehacks.cfg. So you can telnet to the device and modify it with vi.

closed 2020-06-12 by HclX