HclX/WyzeHacks #21: "Configuration file not found, aborting"

chaoticchoas, 2020-05-03:
So I was updating my cameras and wyzehack where uninstalled. I still have access to telnet and I was wondering if I could reinstall the hack without having to take down my cameras. I still have access to telnet and I was able to wget the files to system folder but I can't seem to finish the install as it can't find my configuration files which causes it to abort the install. Is there a way to get around this and then later re insert the configuration files. 

This is what my console shows when running the install.sh

```
wyze_hack/
wyze_hack/app_ver.inc
wyze_hack/auto_archive.sh
wyze_hack/auto_reboot.sh
wyze_hack/bin/
wyze_hack/bin/audioplay
wyze_hack/bin/blkid
wyze_hack/bin/dummy_mmc.ko
wyze_hack/bin/mount
wyze_hack/bin/reboot
wyze_hack/bin/uevent_recv
wyze_hack/bin/uevent_send
wyze_hack/bind_etc.sh
wyze_hack/hack_ver.inc
wyze_hack/hook_init.sh
wyze_hack/install.sh
wyze_hack/log_sync.sh
wyze_hack/mount_nfs.sh
wyze_hack/playwav.sh
wyze_hack/run.sh
wyze_hack/snd/
wyze_hack/snd/begin.wav
wyze_hack/snd/failed.wav
wyze_hack/snd/finished.wav
Starting wyze hack installer...
Enabling telnetd...
Current Wyze software version is 4.9.5.111
Installing WyzeHacks version 0.3.00
Configuration file not found, aborting...
```

chaoticchoas, 2020-05-05:
So I was able to somehow bypass the Configuration file not found, aborting... error. Though I'm not sure how. I'm now able to successfully finish running the install.sh and the device reboots but it seems to not have successfully installed the hack. From a quick look it seems like I'm still missing the wyze_hack folder in /system directory. 
Here is my install log. which I was able to retrieve after changing the location in install.sh
```
Starting wyze hack installer...
Enabling telnetd...
Current Wyze software version is 4.9.5.111
Installing WyzeHacks version 0.3.00
Copying configuration files...
Copying wyze hack binary...
Hooking up boot script...
Done, reboot in 10 seconds...
```

HclX, 2020-05-06:
the latest version changed to a different location, which may be something why you are seeing many file not found errors. To do a telnet installation you need to place the wyze installation files into a directory, together with a config.inc file in the same directory and then run install.sh.

chaoticchoas, 2020-05-07:
@HclX  Thanks after some looking and some confusion I was able to successfully reinstall wyzeHcLX and mount my nfs. I am not able to run ./mount_nfs.sh  without getting the errors mentioned above but restarting the camera seems to correctly mount the nfs without any issues.

If anyone is looking for the new wyze_hack directory it's ```/tmp/run/wyze_hack```. The config.inc file is now called the wyze_hack.cfg and it's located at ```/params/```

closed 2020-05-07 by chaoticchoas