HclX/WyzeHacks #23: Stream quality suggestion

Kotove, 2020-06-04:
Hi, I've noticed there is a dev trace left from tuning:

Kotove, 2020-06-05:
```
typedef struct {
uint32_t maxQp; /**< The encoder supports the maximum QP of the image */
uint32_t minQp; /**< The encoder supports the minimum QP of the image */
uint32_t staticTime; /**< code rate statistics time, in seconds */
uint32_t maxBitRate; /**< encoder output maximum bit rate, in kbps */
     int32_t iBiasLvl; /**< Adjust I frame QP to adjust the image quality of I frame and its code stream size, range: [-3,3] */
uint32_t changePos; /**< The ratio of the bit rate when starting to adjust Qp to the maximum bit rate, the value range: [50, 100] */
     uint32_t qualityLvl; /**< video quality minimum level, range [0-7], the lower the value, the higher the image quality, but the larger the bitstream. minBitRate = maxBitRate * quality[qualityLvl], where quality[] = {0.8, 0.7, 0.6, 0.5, 0.4, 0.3, 0.2, 0.1} */
uint32_t frmQPStep; /**< QP change step size between frames */
uint32_t gopQPStep; /**< QP change step size between gop */
bool gopRelation; /**< whether gop is related */
} IMPEncoderAttrH264Smart;
```

HclX, 2020-06-12:
Thanks for the tip. Since this hack still uses the stock wyze firmware, any increase of the video quality will also increase the size of files being uploaded to wyze's cloud storage, which can increase their operation cost.

closed 2020-06-12 by HclX