HclX/WyzeHacks #25: WyzeHacks does not install

Nu11u5, 2020-07-10:
This is the first time I am attempting to install WyzeHacks on my cameras.

WyzeCam2
FW: 4.9.6.104 (beta)

I have followed the install procedure.  A 32 GB SD card is formatted for FAT32, and the only files on the root are: 

* FIRMWARE_660R.bin
* version.ini
* config.inc (configured)
* install.sh

When the SD card is inserted after a few seconds the camera will chime twice (I believe this is the normal "card inserted" notification).  After that there is never a voice announcement.  If I go to the camera in the Wyze app it lists the SD card as available and in use for local recording.  If I then eject the card and check it there are new folders for the recording files but no log or other changes.

I have attempted to install WyzeHacks on several of my cameras with the same results.

---

*Other things to note:*

The cameras are running beta firmware.  It has recently started frequently losing connection with Wyze servers (app says they are offline but it can successfully directly connect to them regardless).  This is fixed for about 24 hours by physically power-cycling them (reboot commend in the app does not work at this point).  This is likely a bug in the beta FW and I will be rolling back, but it is not obvious if the incompatibility with the WyzeHacks install is due to a bug or an intentional change in the FW.

I already rolled back the beta Wyze app, as it was frequently freezing when opening camera settings.

Nu11u5, 2020-07-10:
Update:

I reinstalled the beta Wyze app and used it to do an OTA firmware downgrade to version 4.9.5.111.  The SD card with the WyzeHacks installer was still in the camera.  During the process the camera rebooted and when the app reported that the firmware downgrade was 20% completed the voice prompt "installation finished, please remove the SD card" alerted.

After the downgrade completed I was able to telnet in but the NFS share was not mounted, even after a camera reboot.  I actually think this may be a firewall issue as I cannot even ping my workstation hosting the NFS server I'm testing with from the camera.  Manually running `wyze_hack.sh mount_nfs` also shows that it can't connect (is this logged somewhere, otherwise?)

Nu11u5, 2020-07-11:
Confirmed that WyzeHacks installs correctly on all of my camera after downgrading to firmware version 4.9.5.111

onfire4g05, 2020-08-05:
Looks like 4.6.9.x has been released. Was this their firmware "rewrite" version? If so, I bet this hack will need changes since they did say they rewrote a bit of code to make better use of the CPU. I wish I knew more about how this stuff works so I could work on the hack.

HclX, 2020-08-06:
i'm aware that the hack will not work on latest firmware (4.9.6.156/4.10.6.156). The new firmware seems to have a big restructuring, which causes the hack no longer working. I'm currently working on that and already got a working version, still in testing for stability before making it public.

onfire4g05, 2020-08-07:
I can try testing, if you want a tester. Is the version in the dev branch working?

HclX, 2020-08-07:
Not yet, it doesn't have the latest changes in my local repo. More
importantly, they blocked the SD card installation method and the telnet
access so I have to find other ways, should be ready in a few days, stay
tuned...

HclX, 2020-08-10:
ok, i have the latest code change uploaded to dev branch. With the latest release (0.4.00) you should be able to get this work on firmware 4.9.6.156 and 4.10.6.156. Keep in mind i just finished this without much of testing, so try it on your risk -- i'd suggest you pick one camera with easy physical access so in case it bricks, you can recover using the SD firmware recovery method.

antontill, 2020-08-10:
Just tried dev branch (0.4.00). Got connected to camera but install failed (camera said "Installation Failed" immediately after beginning installation. Any way to get debug log with remote install?

HclX, 2020-08-10:
Currently the only explicitly failure reason is configuration file not
found, make sure you have a "config.inc" created in the unzipped release
directory (where you run remote_instaol.sh).

There is some logging to SD card mechanism but I just found it won't work
because they changed the SD card mounting point.

antontill, 2020-08-10:
config.inc is in unzipped release folder and I've modified the export line to my NFS location. Strange that it keeps failing during install. I'll keep tinkering around. Running on a Mac btw, don't think that should be an issue though. Thanks for quick reply!

antontill, 2020-08-10:
Antons-MacBook-Pro:wyze_hacks_0_4_00 Anton$ ./remote_install.sh 
Found local config file, including into firmware update archive...
a ./config.inc
tar: --xform=s,./,Upgrade/,g: Cannot stat: No such file or directory
tar: Error exit delayed from previous errors.
INFO:root:Using saved credentials from .tokens...
INFO:root:Checking device, mac=2CAA8EA2E133

Device type:      Camera (WYZEC1-JZ)
Device name:      Front Door Cam
Firmware version: 4.9.6.156
IP Address:       192.168.1.36

Pushing firmware to this device? [y/N]:

HclX, 2020-08-10:
That tar error seems suspicious to me. We are basically appending the
config file to the tar archive in a special folder but looks like that one
failed for some reason. Try to see if you can get that error away.

HclX, 2020-08-10:
I think "--xform=s,./,Upgrade/,g" is not supported by the tar version from MacOS. Can you confirm with the following command from the release dir:

cp ./FIRMWARE_660R.bin ./firmware.bin
tar -rvf ./firmware.bin ./config.inc --xform=s,./,Upgrade/,g
tar -tvf ./firmware.bin

If it works, there shouldn't be any errors and you should see "Upgrade/config.inc" from the last line.

antontill, 2020-08-10:
seems you are correct, I get an error from the 2nd line above. Guess tar Mac version does not support that command. Thanks for helping to debug. I will look into doing this on bootcamp with windows

HclX, 2020-08-10:
Don't bother trying windows, I just found the python coming with wsl on
windows has another compatibility issue with the code. Give me some time
band I should be able to fix both issues tonight.

antontill, 2020-08-11:
I was unable to troubleshoot with any luck at the moment. I'll sit tight and wait for a reposes from you. Thanks!

HclX, 2020-08-11:
@antontill I made another push trying to fix the issue you ran into. Can you give it another try? I don't have MacOS so no way to verify it myself.

antontill, 2020-08-11:
Installation completed successfully on the cam this time! Files don't seem to be uploading to NFS at the moment but that's likely on my end. Thanks for the help!

antontill, 2020-08-11:
In the Wyze app, Advanced Settings>Local Storage, it shows that I don't have a microSD card installed. Is this the expected behavior once telnet is loaded? My cam seems to be pinging the NFS but no bytes written.

HclX, 2020-08-11:
No, that's not right. Try telnet into the camera and run mount to see if
you have the NFS share mounted. I remember someone said they have issue
mounting NFSv4.

HclX, 2020-08-11:
@antontill I just made another push adding a new trouble shooting section about the possible NFS mount issues. I also added another NFS_OPTIONS entry in the config file to make the mounting customizable. take a look and see if that helps.

antontill, 2020-08-11:
@HclX Do you know default Login and Password for the cam? I'm not able to login using telnet.

HclX, 2020-08-11:
For CamV2 the password is "ismart12". For CamPan the password is unknown.
However, if you enables the password shadow option in wyzehacks config, it
will be modified to "ismart12" as well.

antontill, 2020-08-11:
@HclX Thanks.. I found it online after some digging. I'm able to telnet in w/out issues. I can modify config as needed. I think my issue may be regarding my NFS drive. I'm using NFS Manager to mount an external HDD on my Mac. The disk format must be HFS but I'm not certain the camera can write to this drive format. I unfortunately don't have a linux to try it on. I might try a virtual box after a bit more debugging.

antontill, 2020-08-12:
@HclX Got it working! Was a permissions issue ultimately on my NFS drive side. I didn't need to use any special config. Thanks for all the help

onfire4g05, 2020-08-12:
Just installed on 6/7 cams and went perfect! Awesome work, man!

johnfoconnor, 2020-08-12:
confirmed the remote_install.sh works fine on osx with python 3.7 after manually installing 'requests' package

Semag, 2020-08-18:
Hey guys - this is awesome stuff!!!!

@HclX  you might put a note in the main branch that the latest software doesn't work with the sd card method - i was poking around for a while until i saw the "issue" that noted that it's broken on the latest firmware.

Also, my linux box around the house only has python 2.7 - so i made some tweaks to make your python script work on python 2.7, fyi.

Semag, 2020-08-18:
@HclX  one more question for you.  

Can some of your python script be used to spoof the phone app to get the live feed from the app via a PC ?  It seems like it's doing something similar.

There's an android app that's called "TinyCam Pro" that people are using to act as a "man in the middle" to fetch the live feed from the wyze servers and then feed them as a web server camera. This way you can view the feed in HomeAssistant or MotionEye, for example.

I'm wondering if your script can fix that and be used to create a component within Home Assistant to fetch the live feed from the Wyze Servers, so that Home Assistant could act as a "viewer" like the app

Nu11u5, 2020-08-19:
I get this when running wyze_updater.py:

```
python3 -m pdb ./wyze_updater.py update \
>     -m WYZEC1-JZ -m WYZECP1_JEF -f ./firmware.bin -p 11808
> ./WyzeHacks/installer/wyze_updater.py(3)<module>()
-> import sys
(Pdb) c
INFO:root:No saved credentials found, logging in with username/password...
Please enter the account name: *****
Please enter the password: *****
Traceback (most recent call last):
  File "/usr/lib/python3.8/pdb.py", line 1704, in main
    pdb._runscript(mainpyfile)
  File "/usr/lib/python3.8/pdb.py", line 1573, in _runscript
    self.run(statement)
  File "/usr/lib/python3.8/bdb.py", line 580, in run
    exec(cmd, globals, locals)
  File "<string>", line 1, in <module>
  File "./WyzeHacks/installer/wyze_updater.py", line 3, in <module>
    import sys
  File "/./WyzeHacks/installer/wyze_updater.py", line 72, in wyze_login
    session_id = rsp['session_id']
KeyError: 'session_id'
Uncaught exception. Entering post mortem debugging
Running 'cont' or 'step' will restart the program
> ./WyzeHacks/installer/wyze_updater.py(72)wyze_login()
-> session_id = rsp['session_id']
(Pdb) rsp
{'errorCode': 5000, 'description': 'Internal Error', 'requestId': 'e96857ea-b0b7-44b9-be88-bf72b0c4234e'}
(Pdb)
```

I noticed that request is to https://auth-prod.api.wyze.com/user/login/sendSmsCode.  I am using an OTP authenticator app with my account, not SMS.  Is this related?

HclX, 2020-08-19:
@Nu11u5 Yes, the script currently only support SMS based 2FA (only because I haven't setup OTP based 2FA so no way to debugging/testing). I will look into that when I got a chance but no promise how long it will take.

HclX, 2020-08-19:
> @HclX one more question for you.
>
> Can some of your python script be used to spoof the phone app to get the live feed from the app via a PC ? It seems like it's doing something similar.
>
> There's an [...]

It surely can. I actually have figured out quite a  lot of the communication protocols. The video feedback, however, is done over the P2P protocol developed by ThroughTek which doesn't have a python implementation. It's still possible but would take quite a lot effort to make it work, and at this moment, due to the pandemic i don't have much bandwidth on this project.

HclX, 2020-08-19:
> Hey guys - this is awesome stuff!!!!
>
> @HclX you might put a note in the main branch that the latest software doesn't work with the sd card method - i was poking around for a while until i saw the [...]

i'm planning to promote these changes to master since multiple people have verified this is working. The README file will be updated accordingly. Stay tuned.

kdvlr, 2020-08-20:
@HclX - Looking forward to this. One of the things I had modified to my install was to restart the camera whenever the mount was lost/not responding (say due to NAS restarting). With the blackouts, that is happening more frequently. Can your new script take care of restarting if the mount is not responding ?

Here's the previous issue for reference - https://github.com/HclX/WyzeHacks/issues/8

kdvlr, 2020-08-20:
I just tried installing and it seems to stay at the ```Press Ctrl+C when all the updates are done...``` step for a long time. 

Debug:
```
Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file '../installer/firmware.bin' as 'http://192.168.100.221:11808/firmware.bin', md5=3557c3b228dca0b7bf5664fafa43835a
reply: 'HTTP/1.1 200 OK\r\n'
header: Access-Control-Allow-Headers: Origin, X-Requested-With, Content-Type, Cache-Control, Accept, authorization, content-type
header: Access-Control-Allow-Methods: POST,GET,OPTIONS
header: Access-Control-Allow-Origin: *
header: Cache-Control: no-cache
header: Content-Type: application/json; charset=utf-8
header: Date: Thu, 20 Aug 2020 16:17:34 GMT
header: Expires: -1
header: Pragma: no-cache
header: Server: Microsoft-IIS/8.5
header: X-AspNet-Version: 4.0.30319
header: X-Powered-By: ASP.NET
header: Content-Length: 171
header: Connection: keep-alive
DEBUG:urllib3.connectionpool:https://api.wyzecam.com:443 "POST /app/v2/auto/run_action HTTP/1.1" 200 171
DEBUG:root:request received, path=/firmware.bin
192.168.100.147 - - [20/Aug/2020 11:17:33] "GET /firmware.bin HTTP/1.1" 200 -
Press Ctrl+C when all the updates are done...
..................................................................................................................
```

HclX, 2020-08-20:
That's expected, because we don't know when the update finishes (it's
possible but I'm too lazy), so you need to manually verify all camera boot
up correctly then you can terminate the script.

kdvlr, 2020-08-20:
I should have been clear. I let it go for half hour, but nothing happened. The camera did not say Installation Started or anything. The telnet is not working. Nothing is being recorded into NFS. Just the dots that keep increasing. 

Is the last step /GET of the firmware ? I verified that I was able to download the firmware manually as well. They are in the same LAN. Not sure what else I can do.

HclX, 2020-08-20:
From your log, the http server did receive a get request from the device
asking for the firmware, so something must have gone wrong after the
firmware image is delivered to the camera. Can you imsert a SD card and try
it again? The installation script will output some information to the SD
card in a file named install.log.

kdvlr, 2020-08-24:
New issue - I was able to flash and my NFS mount shows up, but it doesn't record anything. I tried turning on and off the recording functionality. I tried formatting it. Permissions are not an issue as it creates all the other files. 

```
192.168.100.45:/export/Cameras/WyzeCams/FrontEntry
                          5.5T      1.5T      4.0T  27% /media/mmc
[root@Ingenic-uc1_1:~]# cd /media/mmc
[root@Ingenic-uc1_1:mmc]# ls -l
-rw-r--r--    1 65534    65534          776 Aug 24 08:29 ifconfig.txt
drwxr-xr-x    2 65534    65534           39 Aug 24 08:29 log
-rw-r--r--    1 65534    65534           20 Aug 24 08:37 time.ini
drwxr-xr-x    2 65534    65534            6 Aug 24 08:29 time_lapse
```

Semag, 2020-08-24:
@kdvlr  did you turn the local storage on in the wyze app?  Does it show the full NFS drive size?

kdvlr, 2020-08-24:
@Semag - Thanks. I did. It shows only 2048G even though I have 5T on my NAS

HclX, 2020-08-24:
2048G is expected: the wyz firmware will not be able to handle larger NFS
shares. To avoid that the new version emulates a total size of 2048G sd
card out of your free space.  I remember there are other issues caused by
either too large NFS share or NFSv4, check the closed issues.

The new version also added configurable mount options (check the readme
file) to get around some special NFS mount parameters.

HclX, 2020-08-24:
> New issue - I was able to flash and my NFS mount shows up, but it doesn't record anything. I tried turning on and off the recording functionality. I tried formatting it. Permissions are not an issue [...]

@kdvlr can you check issue 19, which may be similar to your case? Unfortunately I don't have an NFS share large enough to test it.

kdvlr, 2020-08-25:
Thanks. I checked Issue #19 and it looks exactly like the issue I am having. I have been using this for a while and haven't had any problems till the latest firmware. 

I haven't resolved it, as my NAS has no way of presenting a smaller size to the camera. Let me look into this further.

johnfoconnor, 2020-08-25:
I'm also seeing some likely NFS inconsistencies on the latest build, which i have never seen on previous ones.

Specifically, I see the camera will randomly(?) decide it does not have an SD card inserted. Im assuming this is because the mount is dropped, but i have not had time to investigate

HclX, 2020-08-25:
@johnfoconnor Yes I noticed that too. It seems somewhere in the firmware it decided the "SD card" has gone bad so it went ahead dismount the share. I haven't locate the exact reason yet, but i'm aware of this issue.

HclX, 2020-08-25:
Thanks everyone for the discussion. Now I've merged the dev branch to master. The latest hack is alive and will work with latest firmware. This issue (as the title says) is considered fixed so i'm going to close this one. I opened another issue to track the "random SD card dismount" issue. Feel free to open new issues if you discover any with the latest release.

closed 2020-08-25 by HclX