HclX/WyzeHacks #28: NFS share being unmounted randomly with release 0.4.00

HclX, 2020-08-25:
I've seen multiple cases that the NFS share being unmounted. There is no specific pattern or anything predictable so far. It seems that the latest firmware will think the "SD card" is not working so it dismounts it.

Semag, 2020-08-25:
@HclX  - thanks for the updates here. As I've played with this, i haven't had an opportunity to play with the older versions.  I'm able to reproduce reboots of my wyze and also reproduced the random dismount by going to the "view playback" and starting to move around.  It seems that the app trying to access the historical files is really tough on it, and it is pretty flakey. Is that the way it's always worked with the NFS?

HclX, 2020-08-25:
Per my experience, "view playback" is always slow and flaky, even with SD card. It's getting worse with NFS share because now you have almost infinite storage space  and there are tons of files the camera firmware has to iterate through. That's why I added the "archive old recordings" feature which moves those recordings out of the camera firmware's view to speed things up. But even with that I noticed things sometime go very wrong and even crash the camera.

Semag, 2020-08-25:
Cool, yeah, that's what i felt too.

I was trying to research some software to see if there was a desktop app that could show clips in a timeline the way Wyze does, but no luck yet.

I did find that if i search for *.mp4 on the NFS, i can drag all the search results into VLC and create a playlist of the clips - that works ok.

At sometime i might work on coding a little gui to show the clips as links on a timeline - the folder structure/file naming is very programmatic and makes it easy to find which date / hour / min the file is.

kdvlr, 2020-08-25:
@Semag - I use a script to merge all the files into a 24 hour file to help me manage all this. [Here's ](https://gist.github.com/kdvlr/3cff941d7096d92b6d1d3bc7ba7d9ae5)a link to my script if your're interested. 

I know there is a way to discard video where there is no motion changes, but I need to figure that out..

Semag, 2020-08-25:
@kdvlr  are you recording continuously?  I'm currently offloading some of the motion detection onto the camera, which has a lot of false positives, but I don't think it has *missed* anything really.  This reduces the network strain and writing frequency. 

I want to get the PIR motion detector through the sense package and see how that improves the camera i have mounted by my front door, right now there are way too many false motion events.

I know that there are software packages (blue iris, motion eye, Camect) that will do AI analysis, but they all seem to require the RTSP streams ( which we know is notoriously unreliable on the wyze cam).

thompatry, 2020-08-25:
> Per my experience, "view playback" is always slow and flaky, even with SD card. It's getting worse with NFS share because now you have almost infinite storage space and there are tons of files the [...]

I have witness this too. I was not sure if it was an NFS issue, a hard drive sleeping issue (I have disabled this now), or a camera hardware issue. I have done some troubleshoot but nothing intensive.

HclX, 2020-08-26:
for those who cares, i added change to automatic reboot the camera when the dismount is detected. the change is merged into dev branch so you guys can try it out: you should hear a "double beep" when the camera reboots because of this. I'm still verifying this change so no idea how reliable it works.

kdvlr, 2020-08-26:
Was the release updated again or is the one I downloaded the latest verion. PS unzip in Linux does not unzip the latest file.
```
unzip "wyze_hacks_0_4_00.zip" .
Archive:  wyze_hacks_0_4_00.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of wyze_hacks_0_4_00.zip or
        wyze_hacks_0_4_00.zip.zip, and cannot find wyze_hacks_0_4_00.zip.ZIP, period.
```

HclX, 2020-08-26:
my latest change is in release 0.4.01 archive, just pushed another minor change to avoid bloating log files.

HclX, 2020-08-31:
not really fixing the root cause, but the nfs dismount detection and reboot seems to be working.

closed 2020-08-31 by HclX