HclX/WyzeHacks #29: No video recordings on NFS share but other files are created

HclX, 2020-08-25:
There are multiple reports that on certain setups the NFS share is mounted correctly, some files are created, however, no video recordings. I suspect this has something to do with the NFS share size but will need more information on this since I don't have a large enough NFS share to test this out.

HclX, 2020-08-25:
@kdvlr Let me know if you can post more information about your setup. I added more logging in latest change so can you try it and show me the log related to this. You will need to enable log syncing, and then search "statfs(path=" in the log to find the related debugging information.  Make sure you mask or erase your credentials.

kdvlr, 2020-08-25:
@HclX - Did you make any changes to the NFS code ? I can see the recordings being created with your latest release.

HclX, 2020-08-25:
Yes. I modified the code to change the emulated SD card max size from 2TB to 512GB, and looks like that's the root cause! I will keep this issue open for a couple more days until you confirm this really fixes the issue!

HclX, 2020-08-31:
Mark this one fixed.

closed 2020-08-31 by HclX