HclX/WyzeHacks #30: Cameras rebooting

HclX, 2020-08-26:
This is a weird one: I noticed when log syncing is enabled the camera reboots quite often. It seems to have something to do with log syncing because the reboot is quite unpredictable and is really hard to notice without log syncing enabled.

revenant-81, 2020-08-27:
Not sure if this is related, but when "Event Recording" is enabled, each time a detection occurs, the camera would reboot and the NFS either doesn't generate a file or leave a partial file. Disabling Event Recording, the camera operates normally and generates the files to the NFS.

HclX, 2020-08-29:
i verified log syncing has nothing to do with the reboot: I replaced log syncing with a very short piece code to leave a mark on NFS share every time the device boots. And after that, I can see that the device still reboots very often. So it must be something else.

HclX, 2020-09-05:
I think i got down to the root cause, however, there is no easy solution for that: The problem is my router and unstable network connection. With stock firmware, a broken link between device and router won't cause obvious issues, the device will try to reset wireless network. However, with WyzeHacks installed, because we write directly to the NFS share, if the network goes down, the system will hang and then eventually crash and cause a reboot. This also explains why enabling log syncing or having frequent motion events will make this issue more obvious: the chances that we are writing to NFS share when the network is down increases in such cases. For my case, it's not because poor signal strength, but because some defect in the router (I'm using ASUS AC68U and from the router log I can see the router disconnect that particular camera quite often).

So, conclusion is: make sure your wifi network is stable, and disable log syncing (disabled by default) to reduce the rebooting.

kdvlr, 2020-09-08:
My problem is the other way  around. I have a AC68U  as well and haven't seen disconnects, but if my Wyze has a problem writing to NAS, it waits forever.

HclX, 2020-10-11:
a very interesting observation on my case is if I simply ping the camera or ping the gateway from the camera, it seems the reboot can be prevented: so it seems that running a ping will prevent the wifi being dropped, not sure what's the magic behind this, maybe some sort of power saving functionality in the wifi chip?

HclX, 2020-10-18:
i've beening running 0.4.02 for many days and it seems this issue is fixed by it with PING_KEEPALIVE option enabled.

closed 2020-10-18 by HclX