HclX/WyzeHacks #31: Install issue running remote_install.sh

Pcpro745, 2020-08-28:
i renamed the config file to config.inc and then i ran ./remote_install.sh and i keep getting an error that says 

cp: cannot stat './FIRMWARE_660R.bin': No such file or directory
Found local config file, including into firmware update archive...
Upgrade/config.inc
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (2.22.0)
python3: can't open file './wyze_updater.py': [Errno 2] No such file or directory

closed 2020-08-28 by Pcpro745

cr08, 2020-10-24:
I see you closed this out. Did you fix it? What was the fix? Seeing this on my end (running in an ubuntu WSL instance with all dependencies installed) and not sure of the cause. Everything looks fine, all files are there and readable.

cr08, 2020-10-24:
Figured it out. For future reference for those who run into this: You need to recursively clone the repo to get the dependent WyzeUpdater repo as well which is what was missing and causing the error. A basic `git clone https://....` won't work. You'll need to do `git clone --recurse-submodules https://...`