HclX/WyzeHacks #33: 4.0.0 no longer remembers Name of Camera on NFS after reboot

thompatry, 2020-08-29:
I have noticed with 4.0.0 that if I rename the folder on the NFS share, it persists till I reboot the camera. After I reboot it, it creates the folder with the MAC again. 

This is to track this issue.

HclX, 2020-08-30:
it memorizes the folder by checking a file named ".mac_<mac>" in the folder. Can you confirm there is such file in the renamed folder? Also, sometimes the camera boots up with a different mac, when that happens, it will use a different folder.

thompatry, 2020-09-06:
It is working now. I am not sure what changed but I played around with it the past few days to figure out what was going out and had no luck.

thompatry, 2020-09-06:
I think I have figured it out. I was making it too complicated. The issue isn't directly with 4.0.0 as far as I know. The issue has to deal with renaming the folder and it having spaces. If it has spaces, it then doesn't find the renamed folder. So it might something simple to fix with some quotes.

thompatry, 2020-09-06:
Easy fix right here. Some quotes and viola.

https://github.com/HclX/WyzeHacks/blob/78ea67af6ae4693e5402a7316b6cd7156236e487/wyze_hack/mount_nfs.sh#L45-L47

HclX, 2020-10-11:
got it, will have this included in next release.

HclX, 2020-10-18:
this is fixed in relase 0.4.02

closed 2020-10-18 by HclX