HclX/WyzeHacks #34: RTSP Server

alexandrepossebom, 2020-09-03:
Hello @HclX, thanks for your work.

I have a small suggestion, adding optional support to the RTSP server in your hack will be awesome.

If the problem is space, you can store it on NFS.

thompatry, 2020-09-18:
I have been playing around with this idea but the big issue is the cpu in the camera is not super powerful. You are going to have some artifacts and stuff. I will keep trying different things and let you know the results. Also running at RTSP through the NFS is not advisable because if you lose any connection to the share, the camera could lock up or crash.

exiva, 2020-10-06:
What about storing the binary on SD card if space is an issue? If theres a file on the SD, enable it? The dafang hacks seems to use https://github.com/mpromonet/v4l2rtspserver for the rtsp server.

thompatry, 2020-10-07:
This is a possibility. It would take a little of work but essentially we could mount the sdcard in a different location, set the parameters from the wyzehacks config file, then run the rstp server.

kiwi-cam, 2020-10-08:
Some corners of the internet: https://www.reddit.com/r/wyzecam/comments/esbfe0/rtsp_firmware_and_the_nfssd_card_hack/ff92czn/?utm_source=reddit&utm_medium=web2x&context=3, claim to have applied this hack to the Wyze RTSP firmware. Is this a viable option?

HclX, 2020-10-18:
I had the impression that RTSP will not work with some Wyze's built-in video feature, and that's why they have the special RTSP firmware?

holocronology, 2020-11-11:
> 
> 
> I had the impression that RTSP will not work with some Wyze's built-in video feature, and that's why they have the special RTSP firmware?

That is currently correct.

n1ckyrush, 2020-11-19:
I made a tool that allows to convert MP4 files from NFS to HLS stream. 
HLS stream can be used instead of RTSP in most cases.
Here is the repo: https://github.com/n1ckyrush/wyzecam-hls
@alexandrepossebom

alexandrepossebom, 2020-11-19:
thank you @n1ckyrush I will try.

closed 2020-12-05 by HclX