HclX/WyzeHacks #38: Adjust Volume of "Network connection lost" speaker output prompt

revenant-81, 2020-10-19:
Hi - As of 0.4.02 there was the addition of a voice prompt when network connection is lost. Is it possible to adjust the volume / remove the prompt? Some cameras are placed indoors and the volume seems high for something that could go off at night.

Thanks!

HclX, 2020-10-25:
Sure thing, i sometimes found that's creepy and annoying too. I've made a change in dev branch. Due to limited time I can spend, i haven't really tried this. So there might be typos or anything (which happened a lot before). Please help me verify them and let me know if you run into any problem.

Thanks

revenant-81, 2020-10-26:
Thanks for adding in the volume variable. It's working great!

n1ckyrush, 2020-11-05:
This is a very important fix, please merge to master.
Works pretty well!

closed 2020-11-07 by HclX