HclX/WyzeHacks #40: MQTT support for detections?

roger-, 2020-10-25:
Is it possible to add a small MQTT client to trigger messages based on detections?

HclX, 2020-11-06:
I can probably *hack* something based on the log file, but it can be messy.

roger-, 2020-11-06:
Do you have an example of a log file?

HclX, 2020-11-07:
if you have log sync enabled you should be able to get the log from your camera. There are too many account specific things in the log to share.

HclX, 2020-11-07:
another more logic approach for this is: when events are detected, they will need to post the event to Wyze's server, this is done by using curl library. Since I'm already hooking some functions using LD_PRELOAD, it's possible that we can hook the curl function calls, and extract the event details. However, I'm just too busy at this moment. You are welcome to take a look into it and submit a pull request if you get it figured out. Please do that in dev branch if you are going to.

HclX, 2021-01-25:
closing this for now.

closed 2021-01-25 by HclX