HclX/WyzeHacks #42: 4.9.6.207 beta Firmware

revenant-81, 2020-11-02:
Updating to 4.9.6.207 beta firmware causes the camera to not detect the NFS connection as an SD card even though the mount is still active and valid.

Downgrading to 4.9.6.199 works fine.

gtxaspec, 2020-11-03:
Confirmed, same fw version v2 cam

HclX, 2020-11-07:
yes, i did a brief look and it seems they made some changes to the bootstraping code, will see if i can get over that.

Pcpro745, 2020-11-24:
@HclX any luck with this? Looks like its happening on the new stable firmware too.

HclX, 2020-11-26:
I've drafted release 0.4.04 to support latest stable firmware (4.x.6.218), please try it and let me know if it works.

unomar, 2020-11-27:
I built 0.4.04 from the dev branch and ran the remote_install.sh, but the cameras don't appear to be pulling the update.  Update from 0.4.03 on master still works though.

HclX, 2020-11-28:
Yes there is a bug causing it not working when built from general linux environment. I've made a fix for that.

gtxaspec, 2020-11-29:
confirm latest firmware working with fixed made to dev branch! thanks

closed 2020-12-04 by HclX