HclX/WyzeHacks #50: SD Card not Available 4.9.6.218

Pcpro745, 2020-11-21:
I have the latest firmware installed on my Wyze cams and i also have the latest hack installed on the cameras but the cameras show SD Card not available when trying to view playback. Telnet shows the share is mounted and it will update the share with the Mac and IfConfig files. The share is around 40TB. I used to work before the update.

HclX, 2020-11-26:
I've drafted release 0.4.04 to support the this firmware, please try it and let me know if it works.

HclX, 2020-11-26:
forgot to mention, it's currently on dev branch, please find the release zip file in the "release" directory of the source code.

gtxaspec, 2020-11-29:
confirming, latest dev branch works great, NFS etc. working good.

thank you!

Pcpro745, 2020-12-02:
Thanks!!! appreciate it.

HclX, 2020-12-04:
This is now in latest release

closed 2020-12-04 by HclX