HclX/WyzeHacks #51: [Solution] Windows NFS Server

twig123, 2020-11-23:
Hello,

After much headache, I've found a solution for hosting a NFS share on Windows and getting Wyze Hacks to work with it. I'm hoping this info could be added to the setup information somewhere.

Grab a copy of haneWIN NFS Server for Windows:
https://www.hanewin.net/nfs-e.htm

Edit the "exports" file, and place a # at the beginning of each line (to comment out the default lines).
Add a new line similar to the `following:

`A:\NFS -name:nfsroot -umask:000 -public -mapall:0`

(Edit/Change "A:\NFS" to the folder you want to be the root of your NFS share.

Run "nfssrv.exe" to start the NFS Server.

For your cam, edit the /params/wyze_hack.cfg and set the following:

`export NFS_ROOT='192.168.1.147:/nfsroot'`
(Edit/Change "192.168.1.147" to the IP of your NFS Server's IP)

`export NFS_OPTIONS='-o nolock -o tcp -o rsize=32768,wsize=32768'`

You should now have a fully functional NFS Share on Windows 😎

closed 2020-12-05 by HclX

gszakacs, 2021-05-17:
Any idea if the camera NFS options can be modified for compatibility with WinNFsd ?

HclX, 2021-05-17:
I have never tried that since I have a dedicated Linux server. However, I
remember someone created a writeup in one of the closed issues.

gszakacs, 2021-06-01:
To the folks who want to record to nfs using windows 10:

After some trial&error, I found that the winnfsd version posted at
https://github.com/winnfsd/winnfsd/issues/73#issuecomment-452092487
resolved all of my nfs issues on windows. This version was  not officially released  so you have to download it from the linked post.

Example:
In the wyzehacks config.inc, set the nfs share:
export NFS_ROOT='192.168.2.43:/mnt/nfs_share'

then call winnfsd-64byte from a batch file with the following switches: 
D:\mnt\WinNFSd-64byte.exe -id 0 0 -addr 192.168.2.43 -log on D:\mnt /mnt/nfs_share

The above allows the reliable recording of the videoclips from 2 x wyzecam v3 and 1 x wyzecam v2 to the nfs share,