HclX/WyzeHacks #53: Wyze Cam v2 v4.9.6.199 No NFS or Telnet on v0.4.04

jdkadel, 2020-11-26:
I really like the idea of storing the files on a NFS mount.  I tried with v 0.4.03 then with v04.04 and both get to the "installation finished" voice and reboot but no NFS mount or telnet access.  I've tested the NFS mount (EXT2, 700GB Avail) from my workstation and it's works from there.  I'm using the default config template with only the NFS line updated.

The install log was created the first time. Which I didn't save :(  Now every install since it hasn't created the install log.

Suggestions?

gtxaspec, 2020-11-26:
there is a dev branch that fixes these issues, stay tuned for a new official release from the author soon

HclX, 2020-11-28:
@jdkadel I hate to say this but "it works for me".  One trick you could try is to check my WyzeUpdater repo, which has a sample update allowing you to enable telnetd for the camera. That should give you telnet  access to the camera. And then you can look around and check what's wrong with last installation. Without that detail, there is not much I can help.

jdkadel, 2020-11-28:
Soon after trying to install I thought I had bricked the camera.  Back at it today and it's back online.  I tried to install v0.3.03 again and it went through the process but still didn't give access to telnet or the NFS mount.  But! it did create the install log.
[install.log](https://github.com/HclX/WyzeHacks/files/5610851/install.log)

When I attempt to connect with Putty Telnet it times out with Network error: connection refused.  If I try with SSH it asks for name and password?  

The NFS mount I'm using is '192.168.1.3:/mnt/HD_a2/Wyze/' No other changes to the config.inc.

HclX, 2020-11-28:
Your "install.log" looks fine to me. Are you sure you are connecting to the right camera with telnet/ssh? sshd is not available on the camera so I don't expect anything asking you username/password when you tried ssh.

I also noticed that your NFS root variable has a trailing slash, not sure if it matters but mine doesn't have one. Maybe you want to remove the trailing slash and try it again?

jdkadel, 2020-11-28:
Basic mistake.  I just used the first IP address displayed.  Telnet works fine if you use the correct IP.  Thanks for the pointer.

The NFS connection is being refused by the host.  The cam and the NFS are on the same network and works on another Linux box.  I'll have see what's up. 

Thanks for the help.

jdkadel, 2020-11-29:
The NFS issue ended up being a rights issue on the NFS server share.  Once changed the cam started writing files.

closed 2020-11-29 by HclX