HclX/WyzeHacks #55: Private key

bettse, 2020-11-28:
I was looking through the firmware, running strings on the iCamera binary and saw "KeyN:" and "KeyE:" followed by a series of hex strings.  I recognized the terms as being related to RSA.  I took the hex strings and with a little python found they are very likely for a private key.  I have no idea what it is used for, but I thought you might have an idea or find it interesting.

-----BEGIN RSA PRIVATE KEY-----
MIICXAIBAAKBgQDhC+pJHrxrZd4OKzI2b8qKUSvswWj39YKe8mBHOsJ1en6kMtg3
Hnoq8Kzv6sihKMeXjEX8k00Z623KJjTMmLQtTUWq6ylKzr/0u5Q4xL12o+rbS7eu
GLyZqVsKoWty08S1LxC4ocXrYJaRYVQIZa7PwWqHKF/FIFKFIda+JEThHwIDAQAB
AoGAWI2Ttwn9IlWHNxQQQhNjI2IlYpgIcemNaHX/JIgelJOK2ZmYMrgvYk4p/o3o
PGILh7qOn4Bmylg1b2HxTMCZxKyt90/lGYzUu8i+jvxvcsvrwejJwVluk4bXhGJC
pw4brXNQzngIBHstPhjfo6aiQXKpK3a+dSL9q1ooxE7CgoECQQD4TkLthpW3oQsd
fceIrszD+sD8vfqTUlrXL8/XBONejmkCmVUMe6WHlMzAHrpCib+jb0K4mznsLOzY
Igk1sm7vAkEA6AUlOPSw3K/hvZfGdXOR+gA+9DSRTPNPKrA4gnVl9H11mXuzEjAj
vwU/8WblUwOgr69xYyk+1IBOJ2+1c+yw0QJBAK31VlpGQbnGb50Xs6UEoZxjnrTy
/AweVFoRuxCu8gQazWLuvXDjZ1KXYuLuJBvdmY8MsbfXuDrINp4tKjqeaa0CQFN4
LmjsFoh+Oc70QDBW0GhJGFzQYIl3ar6nwNymEXTAgcMirkxXxzRcViGpa8zOnEs3
6anL18+pDPTBCl1XBFECQHcXY2P8emOGbZzI6Mz2nEOwV3HdQ0YzNZHlF7yqtVtv
ky3Air5FKHWoFNu1pOGwTxyllxMCwhP0EtBY3SHHPvk=
-----END RSA PRIVATE KEY-----

Just a note: I'm fairly confident the numbers I found are a private key since the python script I used didn't require inputting the exponents, but if you run `openssl rsa -text -in wyze_private.pem`, the exponents match.  Meaning the hex strings are internally consistent with those stored in a private key struct.

HclX, 2020-11-28:
What version of firmware are you talking about?

bettse, 2020-11-28:
Ah, sorry.  It was the rtsp one from the wyze forums.  Comparing sha1sum, I think it is demo_V2_RTSP_4.28.4.41.bin

HclX, 2020-11-28:
interesting, i do see a function call to rsa_private_decrypt with hardcoded private key bytes, however, i couldn't find any code referencing this function.  This can be a security problem, do you want to report this to wyze? I can help connect you to some wyze employees for faster response.

closed 2020-12-05 by HclX