HclX/WyzeHacks #56: working with 4.9.6.224

gtxaspec, 2020-12-05:
updated to 4.9.6.224, can confirm this works well.  

Anyone else can confirm as well?

HclX, 2020-12-05:
Thanks, please keep doing this for new future releases and let me know when it's working or not.

gtxaspec, 2020-12-06:
works with RTSP firmware version: 4.28.4.49 as well

revenant-81, 2021-01-14:
I ran into an issue where upgrading V2 to 4.9.6.232, it lost telnet. Attempting to remote install didn't work. I performed a factory reset back to 4.9.6.218 and attempted to install the hack using remote install, however it seems the file isn't being pushed to the camera even though is shows "Serving firmware file './firmware.bin' as http://<pc ip>:11808". "installation begins" is not triggerred.

gtxaspec, 2021-01-15:
4.9.6.232 works

TCSFrank, 2021-07-09:
Doesn't seem to work on 4.9.6.241 v2

MomosX, 2021-07-12:
Confirming it works on V3 FW 4.36.25 :)  thank you for the great work !!! 
Tested on 4 cams.
My setup is 4 x Wyze V3 , NFS share on TrueNAS 12.0 U-4

Setting up the NFS share the right way took me a long time, as i was getting video files with 0 bytes. Problem was the ACL Mode on the Pool which was set to Restricted instead of Passthrough.