HclX/WyzeHacks #58: Firmware for WyzeCam v3

gtxaspec, 2020-12-06:
https://download.wyzecam.com/firmware/v3/demo_wcv3_4.36.0.213.bin.zip

https://download.wyzecam.com/firmware/v3/demo_wcv3_4.36.0.125.bin.zip

perhaps these will be helpful to investigate the viability on WyzeHacks working on the V3 model.

HclX, 2020-12-07:
I've got the v3 camera and managed to dump the firmware out. It's a little bit tricky with v3: they changed the appfs to a readonly squashfs format, which means we won't be able to install anything onto it, which is how we bootstraping wyzehacks currently. One possible approach is to do a onetime rootfs modification so it can load wyzehacks.

Anyway, thanks for the information. It will be handy in case I bricked my v3 camera during experiment.

HclX, 2020-12-13:
@gtxaspec have you tried sd card firmware recovery with these firmware images? I think I just bricked my v3 and sd card recovery doesn't work somehow. I guess it's time to tear it apart and solder serial wires...

gtxaspec, 2020-12-13:
Negative, not yet.  I have 4 v3s on order, get to receive them.  Once I do, i will👍 I don't mind cracking some open and bricking a few of them in the name of research lol.

Maybe try renaming the file? Did you dump the bootloader? I wonder if maybe the file name changed... If you did dump the BL maybe the file name is buried in there.

I will definitely hook up serial when I get mine... I see some spots for pin headers on the fcc internals photos..

HclX, 2020-12-13:
@gtxaspec nevermind, according to their official website, for v3 cam you need to name the file to "demo_wcv3.bin" instead of "demo.bin", and I verified that works. I'm still having trouble getting the camera accept my image file containing a modified rootfs image, but that's a different issue.

gtxaspec, 2020-12-13:
@HclX great to hear the SD card restore works, last time I checked the official site there was no mention of the official firmware links, or the file name, nice to see they updated their documentation.

hope you can get the rootfs mod working! hope I can get my cams soon =D

gtxaspec, 2020-12-14:
T31 (wyze) tidbits:

u-boot: https://github.com/bakueikozo/atoms-uboot
kernel: https://github.com/bakueikozo/atoms-kernel

revans23, 2021-01-08:
+1 for v3 support.  I just got two of these units in yesterday and am eager to get the hack working on them.

HclX, 2021-01-18:
I got something semi-working on v3 camera, and with a catch: you will have to manually flash something using an acquired telnet shell and it has high chance of corrupting the camera's flash (which can still be recovered with SD card method). If anyone is willing to give it a try, I will come up with a small write up and the related tools.

cheesefinger, 2021-01-18:
@HclX I have time for testing

revans23, 2021-01-19:
With my friend’s help I can test.

HclX, 2021-01-19:
For those who want to try this out, please use the latest release (0.5.01) in dev_v3 branch, and follow steps in https://github.com/HclX/WyzeHacks/blob/dev_v3/tools/v3/README.md. Keep in mind that this hasn't been fully tested yet and it could brick your device. Please read through the entire document before you proceed and do it at your own risk!

HclX, 2021-01-19:
please also test this release with v2 and pan camera if you can, it fixes a bunch of things and restructured the code quite a lot (due to v3 support).

revans23, 2021-01-19:
I ran the special steps to allow the hack to be loaded on two cams and received:

> Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './wcv3_init.bin' as 'http://192.168.1.66:11808/firmware.bin', md5=a006e7fc8b9001e6d83678f9597e9b38
192.168.1.10 - - [19/Jan/2021 14:03:51] "GET /firmware.bin HTTP/1.1" 200 -
INFO:root:Checking device, mac=2CAA8EF31F82

> Pushing firmware to this device? [y/N]:y
192.168.1.135 - - [19/Jan/2021 14:03:59] "GET /firmware.bin HTTP/1.1" 200 -

Then I tried to install the hack and received:

> Pushing firmware to this device? [y/N]:y
INFO:root:Serving firmware file './firmware.bin' as 'http://192.168.1.66:11808/firmware.bin', md5=72209ba3f671b378e6dba498faca03f9
192.168.1.10 - - [19/Jan/2021 14:13:42] "GET /firmware.bin HTTP/1.1" 200 -
INFO:root:Checking device, Mac=2CAA8EF31F82

> Pushing firmware to this device? [y/N]:y
192.168.1.135 - - [19/Jan/2021 14:13:47] "GET /firmware.bin HTTP/1.1" 200 -
INFO:root:Checking device, Mac=2CAA8EA00AB8

However neither is able to see the NFS share.

HclX, 2021-01-19:
@revans23 can you confirm the firmware version of the device? You need the version "4.36.0.228" to begin with.  Can you also try to telnet into those cameras to see if telnetd is enabled or not?

revans23, 2021-01-19:
@HclX Yes, they are both running the appropriate firmware.  The telnet connection is refused from both v3 cams, but I am prompted to log in when attempting to connect to my v2 cameras.

HclX, 2021-01-19:
That means something bwent wrong with the first step. Can you check if
there is a log file on the sdcard? It should be something like v3_init.log.

revans23, 2021-01-19:
I did not use a SD card because the cameras already had the correct firmware installed.

gtxaspec, 2021-01-20:
tested on v2, works on .218, bricks (yellow solid light forever) on .232.

HclX, 2021-01-20:
@revans23 did you verify the camera reboots correctly after the first update? put a sd card in and retry the process, there may be some logs available.
@gtxaspec 232 is pre-release, so no guarantees, it's probably due to a changed init script. Have you tried on v3 camera?

cheesefinger, 2021-01-20:
Worked on the first attempt. 

> The init process will also enable telnetd and change the password to ismart12 so you can telnet into it even without wyze hacks installed.

This did not happen for me. Password remained WYom2020

Logs attached.

> Device type:      Camera (WYZE_CAKP2JFUS)
> Device name:      Lab
> Firmware version: 4.36.0.228
> IP Address:       192...
> 
> Pushing firmware to this device? [y/N]:y
> INFO:root:Serving firmware file './firmware.bin' as 'http://192...:11808/firmware.bin', md5=80a7195653a8892f1b57873dfe73d350
> 192... - - [20/Jan/2021 09:23:57] "GET /firmware.bin HTTP/1.1" 200 -
> INFO:root:Checking device, mac=2Cxxxxxxxxxx

> Trying 192.......
> Connected to 192....
> Escape character is '^]'.
> 
> WyzeCam-xxxx login: root
> Password: 
> Login incorrect
> WyzeCam-xxxx login: root
> Password: 
> [root@WyzeCam-xxxx:~]# ps
>   PID USER       VSZ STAT COMMAND
>     1 root      1600 S    {linuxrc} init
>     2 root         0 SW   [kthreadd]
>     3 root         0 SW   [ksoftirqd/0]
>     4 root         0 SW   [kworker/0:0]
>     5 root         0 SW<  [kworker/0:0H]
>     6 root         0 SW   [kworker/u2:0]
>     7 root         0 SW   [rcu_preempt]
>     8 root         0 SW   [rcu_bh]
>     9 root         0 SW   [rcu_sched]
>    10 root         0 SW   [watchdog/0]
>    11 root         0 SW<  [khelper]
>    12 root         0 SW<  [writeback]
>    13 root         0 SW<  [bioset]
>    14 root         0 SW<  [kblockd]
>    15 root         0 SW   [khubd]
>    16 root         0 SW   [kworker/0:1]
>    17 root         0 SW<  [cfg80211]
>    18 root         0 SW<  [rpciod]
>    19 root         0 SW   [kworker/0:2]
>    20 root         0 SW   [kswapd0]
>    21 root         0 SW   [fsnotify_mark]
>    22 root         0 SW<  [nfsiod]
>    23 root         0 SW<  [crypto]
>    37 root         0 SW   [kworker/u2:1]
>    38 root         0 SW   [kworker/u2:2]
>    39 root         0 SW<  [deferwq]
>    40 root         0 SW<  [kworker/0:1H]
>    52 root         0 SWN  [jffs2_gcd_mtd6]
>    53 root      1632 S    sh /tmp/wyze_hack/run/main.sh run
>    54 root      1600 S    /sbin/getty -L console 115200 vt100
>    94 root         0 SW   [irq/37-isp-m0]
>    96 root         0 SW   [irq/38-isp-w02]
>   132 root         0 SW   [mmcqd/0]
>   162 root     18612 S    /system/bin/hl_client
>   163 root      594m S    /system/bin/iCamera
>   168 root         0 SW   [ksdioirqd/mmc1]
>   171 root     64520 S    /system/bin/assis
>   212 root      1744 S    {sysMonitor.sh} /bin/sh /system/bin/sysMonitor.sh
>   233 root         0 DW   [isp_fw_process]
>   280 root         0 SW   [RTW_XMIT_THREAD]
>   281 root         0 SW   [RTW_CMD_THREAD]
>   282 root         0 SW   [RTWHALXT]
>   347 root      5384 S    wpa_supplicant -D nl80211 -i wlan0 -c /tmp/wpa_supplicant.conf -B
>   399 root      1756 S    udhcpc -i wlan0 -p /var/run/udhcpc.pid -b
>   479 root     15852 S    /system/bin/kvs_stream
>   485 root      1592 S    telnetd
>   828 root      1584 S    sleep 60
>   841 root      1600 S    -sh
>   911 root      1736 S    sleep 2
>   912 root      1592 R    ps

[v3_init.log](https://github.com/HclX/WyzeHacks/files/5843503/v3_init.log)
[install.log](https://github.com/HclX/WyzeHacks/files/5843506/install.log)

HclX, 2021-01-21:
nice, thanks for verifying @cheesefinger.  I'm integrating `dev_v3` back to `dev` branch.

jk111, 2021-01-22:
I was able to telnet into my v3 as well after running the init script. After running the remote_install script though, I'm not seeing any SD card inserted in the Wyze app. I also don't see any traffic from my cam to my NFS box.

HclX, 2021-01-22:
@jk111 can you confirm your firmware version? I know that the latest beta version firmware doesn't work with the hack.

jk111, 2021-01-22:
@HclX I'm running 4.36.0.228.

jk111, 2021-01-22:
@HclX Ignore... user error :)

Working great now.

HclX, 2021-01-22:
Awesome! Make sure you don't install beta firmwares because I know they
changed something causing sd card emulation not working:)

revans23, 2021-01-22:
@HclX I have the cams mounted in such a way it is hard to access them to get a SD card back into them.  Is it possible to change the logs to write to where I am running the installer from?

HclX, 2021-01-22:
Well, it's v3 so I assume you don't have nfs mounted. Not sure where else
you can run the v3_init installer from.  Once you done v3 init, you should
have telnet access through which you can manually mount a NFS share
containing the installer. When that's done, you can always bind the nfs
share as /media/mmc and that will allow the installer to write logs into
your nfs share.

closed 2021-01-26 by HclX

revans23, 2021-01-26:
I have managed to verify telnet access to the cam.  When I run remote_install in Terminal it goes through the necessary responses, cam reboots, but it does not see the NFS share to record to.

revans23, 2021-01-28:
> I have managed to verify telnet access to the cam. When I run remote_install in Terminal it goes through the necessary responses, cam reboots, but it does not see the NFS share to record to.

Fixed it, my release was garbled up.  Installed on one v3, working on the other now.  Thanks for making this happen @HclX !

hyukishi, 2021-02-01:
> @HclX Ignore... user error :)
> 
> Working great now.

@jk111 What was your user error? I'm still having problems getting my v3 to allow telnet access beyond the v3_init.sh script. After running remote_install.sh I don't have telnet access any longer.