HclX/WyzeHacks #59: Having issues installing  -TypeError: __init__() got an unexpected keyword argument 'required'

jakepals, 2020-12-08:
I recently downloaded the WyzeHack and I try to run it, but below is my roadblock. 
What am I missing?

` ./remote_install.sh
`
```
Found local config file, including into firmware update archive...
Upgrade/config.inc
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: requests in /home/akiva/.local/lib/python3.6/site-packages (2.25.0)
Requirement already satisfied: certifi>=2017.4.17 in /home/akiva/.local/lib/python3.6/site-packages (from requests) (2020.12.5)
Requirement already satisfied: chardet<4,>=3.0.2 in /home/akiva/.local/lib/python3.6/site-packages (from requests) (3.0.4)
Requirement already satisfied: idna<3,>=2.5 in /home/akiva/.local/lib/python3.6/site-packages (from requests) (2.10)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/akiva/.local/lib/python3.6/site-packages (from requests) (1.26.2)
Traceback (most recent call last):
  File "./wyze_updater.py", line 294, in <module>
    help='Supported actions')
  File "/usr/lib/python3.6/argparse.py", line 1716, in add_subparsers
    action = parsers_class(option_strings=[], **kwargs)
TypeError: __init__() got an unexpected keyword argument 'required'
```|

HclX, 2020-12-08:
it appears you have python 3.6 installed but the installer depends on something available only since 3.7. Easiest solution is to install python 3.7, or you can modify the installer code a little bit if you know python.

jakepals, 2020-12-08:
I did have 3.7 installed, as well as 3.6 (was automatically installed when I install Samba). 
However I didn't realize 3.6 was default. So I changed it 
`sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 2`
`sudo apt-get install python-pip python-dev libssl-dev libcurl4-openssl-dev libjpeg-dev libz-dev python3-pip`
and it worked. Thank you.

closed 2020-12-08 by jakepals