HclX/WyzeHacks #60: New hardware?

endertable, 2020-12-08:
The software works perfect on several of my cams using the SD card method that have FCC ID ending in C2, but I noticed new hardware with FCC ID ending in C02, it does not want to read firmware off the SD card anymore?  I have used several WYZE firmware bundles to try it with: 
4.9.5.115 
4.9.6.156
4.9.6.199
RTSP 4.28.4.49 (because the prior RTSP 4.28.4.41 does not work with these new cams either.)

I would be happy just getting telnet enabled as I really don't use the NFS at this time as I don't have NFS on my server so I use a script to use ssh to copy over to my server which works well enough.

Have you encountered this new hardware yet?

HclX, 2020-12-08:
Is it possible a v3 camera? First, the latest firmware (I forgot since when) no longer read update files from SD card (which is why I had a massive change on the installation method, check README file). Second, the v3 camera is different than v2 as it no longer allows write operation to the appfs, which prevents wyzehacks from being installed.

If you just want telnet access, check my other repo WyzeUpdater, which has a sample firmware (wyzecam_telnet.bin) enabling one-off telnetd support. Once enabled, you can telnet into the camera and make it persistent (as long as it's not v3 camera).

let me know if you run into any issues.

endertable, 2020-12-08:
It is the V2 black picked up at home depot with a 32GB SD card for 27.88 right now.  Mind you, I always used your SD card install and it worked perfect on the other white cams that I also picked up at HD but they came in a WYZE Smart Home Starter Pack for 49.99.  I tried using the python script but because I also have python 2.7, that I can't uninstall, it creates alot of issues forcing it into the python 3.7 environment.  (I'm obviously not handy with python)  But I'm just going to have to spend up to a few hours with my Linux system again it and I will let you know how it goes.

endertable, 2020-12-09:
OK, well now a big victory just in the the python department!!  Installed virtualenv then rebuilt p3.72 from source to include SSL to be able to get 'requests' module.  Ran wyze_updater, fed credentials and retrieved lists of my cams.  All good, until I run:

./wyze_updater.py update -p 18080 -d 2CAA8EB9DA7B -f $PWD/camera_telnet.bin

It just keeps building dots across the screen.  I know the file is serving because I can use wget and retrieve it from another machine:

wget "http://server1:18080/firmware.bin"

And I can see that GET request on the serving machine, but the cam is not requesting.  I can ping camera from serving machine as well.  Is there more I can add to command line to force camera to get it
Thanks

endertable, 2020-12-09:
OK, while I was showering I was pondering and i remembered that when you have RTSP software installed (which I did), as far as the camera/app is concerned, there is no update firmware available, I imagine same for the USB webcam firmware, and of course if you already have the latest firmware (varies to whatever it is ATM).  So I downgraded FW back down to v4.9.5.115 and voila, your script instantly worked!!  So people that are just seeing dots and their cams not making GET requests, having one of those 3 firmwares currently installed could be the reason...  Just FYI

gtxaspec, 2020-12-10:
the latest official, as well as the latest rtsp firmware (updated about a week ago) are confirmed to work!  (not sure about the old rtsp firmware)

endertable, 2020-12-11:
> 
> 
> the latest official, as well as the latest rtsp firmware (updated about a week ago) are confirmed to work! (not sure about the old rtsp firmware)

Were you able to get telnet on the new RTSP firmware, demo_v2_rtsp_4.28.4.49.bin as that is the problem I was having.  Thanks

gtxaspec, 2020-12-12:
I'll go pick one of these kits from home depot and report back tomorrow

endertable, 2020-12-12:
OK, I did something really cray cray.  I tool all the /system/bin binaries from RTSP firmware (demo_V2_RTSP_4.28.4.41.bin) and copied them over stock firmware (demo_4.9.5.115.bin) as well as missing libraries, differing config files, etc.  

Status:
RTSP:  _**working**_
Telnet:  **_working_**
Record to SD card:  **_working_**
Cam as AP:  **_working_**
App changes settings:  **_working_**
App view cam video:  **_not working_**

It's strange as the App sees the cam "On" and can modify cam settings and you immediately see them take place so I know the cam is connected to the cloud but I cannot get the video feed.  I get error code 90.  I kind of like the all-in-one aspect of the cam, e.g., SMART cam (stock mode), RTSP stock cam, cam as access point and can view live RTSP without needing wifi router or switch.
The only one I lost in this craziness is the SMART cam (stock) view.

The other point to the author or anyone else, is there a way to send message to iCamera to tell it to flip screen.  I notice the uevent_send binary to let iCamera know SD card inserted but can it be used to tell iCamera other things as well, e.g. flip the screen, turn off LEDs, etc.  

Thanks.

gtxaspec, 2020-12-14:
just picked up a kit from home depot with FCC ID ending in C2, wyze hacks works well, both on stock, AND rtsp firmware.  

used SD card to flash RTSP firmware, then used remote install for wyze hacks.

FYI.

endertable, 2020-12-14:
> just picked up a kit from home depot with FCC ID ending in C2, wyze hacks works well, both on stock, AND rtsp firmware.
> 
> used SD card to flash RTSP firmware, then used remote install for wyze hacks.
> 
> FYI.

Yep, that series has always worked for me, too.  Did you buy a balcony one.  They usually have the new FCC number and newer hardware

gtxaspec, 2020-12-14:
@endertable my mistake, I had meant, FCC ID ending in "**C02**"

So i can confirm it works on both. C2/C02

endertable, 2020-12-15:
Hi I picked up another black, sinCe they’re so cheap.  It is the new version so can you tell me the exact firmwares, the order you loaded, and when you used wyze hacks and which version
If you do things in a bad order, wyze hacks won’t work

gtxaspec, 2020-12-15:
I didn't pay attention to what firmware came on the unit...

1.  Put demo.bin ( demo_v2_rtsp_4.28.4.49.bin)
2.  With unit powered down, put micro sd card in, hold setup button, then plug power in
3. wait a bunch, and pull the sd card out, setup camera, add to account via app
4.  use remote_install.sh to flash to camera, with config.inc setup prior with nfs mount, etc options.
5. camera plays sounds, reboot, and telnet in to verify nfs
6. verified everything works.

i flashed rtsp and stock rom about 6 times total comparing between, both firmware worked fine.

holocronology, 2020-12-15:
Just to be clear, I think the Black Cam has always been a different hardware version. I had one from back in 2018 and it's been nothing but trouble with firmware updates. The regular v2 never had an issue.

endertable, 2020-12-16:
> 
> 
> Just to be clear, I think the Black Cam has always been a different hardware version. I had one from back in 2018 and it's been nothing but trouble with firmware updates. The regular v2 never had an issue.

And it still continues to be, for me at least.  And it's too bad because HD has them at such a good price with a Wyze SD card.   But sure enough, Wyze hacks does not want to work with the RTSP firmware  ( demo_v2_rtsp_4.28.4.49.bin) installed, for me anyway.  But I can still install stock firmware 115 and use the kludge method I did above for RTSP.

gtxaspec, 2020-12-16:
Something in your process or software must be off... All hardware versions work fine for me. 

What OS are you using to install?  What kind of router do you have?

endertable, 2020-12-17:
I tried it again today after downgrading to 115 firmware then back to the RTSP 49 and it magically worked.  Now to make sure I don't lose it  :)

endertable, 2021-01-03:
Closing issue

closed 2021-01-03 by endertable

revenant-81, 2021-02-08:
Hi @endertable, I'm facing the same situation where I'm unable to successfully install Wyzehack over RTSP 49. I've attempted the stock flash and downgrade to 115 then flashed RTSP 49 over it and have had no luck so far running the remote install of the hack.

Mind if I ask what steps you took to finally get the hack installed on RTSP 49?

Thanks!

HclX, 2021-02-08:
Hey, when you say it failed to install what is the exact symptom? Does it
say installation failed? Or it succeeded but sdcard won't show up in the
app? There are version checkings in the installer which might be a reason.

revenant-81, 2021-02-08:
I think I may have solved it. 
0.5.01 - Was giving me "Installation Failed".

I figured to give 0.4.04 a shot and installation completed and NFS mounted.

HclX, 2021-02-08:
I see. That must be caused by the version check I added. What's the
firmware version the camera is running on?

revenant-81, 2021-02-08:
It's Wyze's official RSTP firmware v49 (demo_v2_rtsp_4.28.4.49.bin)

HclX, 2021-02-09:
@revenant-81 I opened a issue [82](https://github.com/HclX/WyzeHacks/issues/82) for this. The fix is made in branch `bugfixes`. Can you give it a try?