HclX/WyzeHacks #61: Auto update seems to lock in settings

lando-cal, 2020-12-09:


lando-cal, 2020-12-10:
So, I can't check much beyond this, as I don't have Telnet access (I didn't set the hash yet, and I don't know the defaults) but in the log, it's printing the existing configuration, rather then the new one, when it shows that it's found a new one. It's doing this when printing what it believes to be the new config, and again once printing what's been applied.

twig123, 2020-12-13:
default credentials are listed in step 6 here:
https://github.com/HclX/WyzeHacks/blob/master/info/sdcard_hack.md

lando-cal, 2020-12-15:
> 
> 
> default credentials are listed in step 6 here:
> https://github.com/HclX/WyzeHacks/blob/master/info/sdcard_hack.md

Thanks for that, not sure how I missed it.
After review, I've noticed that the configure function hooks the reboot, not startup. Rebooting the camera from the app or Telnet works fine. If it loses power and comes back online, however, this will overwrite any config intended to be retrieved by the camera.

HclX, 2021-01-26:
this is fixed in release 0.5.01

closed 2021-01-26 by HclX