HclX/WyzeHacks #62: Request for USB Ethernet driver?

ryny24, 2020-12-12:
I just found this WyzeHacks and I can't wait to try it.

Would it be possible to add ethernet (USB) support to the WyzeCam? For locations that don't have Wifi. Just a thought!

Keep up the good work!

gtxaspec, 2020-12-12:
I was just thinking of this before bed last night, lol.

Reading the source to dafang, you need to compile usb-net support for the kernel and probably an asix ethernet device driver, as modules.  

Can these be used from the dafang source? Hmm

gtxaspec, 2021-01-31:
Download:

https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/driver/usbnet.ko
https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/driver/asix.ko

copy these onto the camera somehow (you could have a custom script do this probably), then run the commands

cd /tmp
insmod usbnet.ko
insmod asix.ko

then plug in your usb to ethernet adapter:

`[44726.370013] usbcore: registered new interface driver asix
[45015.713488] usb 1-1: new high-speed USB device number 2 using dwc2
[45016.558155] asix 1-1:1.0 eth0: register 'asix' at usb-dwc2-1, ASIX AX88772B USB 2.0 Ethernet, xx:xx:xx:xx:xx
`

ryny24, 2021-02-01:
NO WAY! No way! I can't wait to try this. Thank you!

gtxaspec, 2021-02-01:
remember, you'll need an ASIX based ethernet adapter; with a specific model ASIX chipset

ref https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/pull/536
https://cateee.net/lkddb/web-lkddb/USB_NET_AX8817X.html

gtxaspec, 2021-02-06:
looks like the V3 cam has these modules built in to the kernel.  I am waiting for a pass through usb otg adapter to arrive and then I can test this on the V3 cam

ryny24, 2021-02-06:
Wow, that's great!  Sadly I don't have any of pass-thru cables but they are on the way now!   I have several V2 on NYC roof tops with a great view. I'll get some swapped out to V3 with ethernet. I'll share the feed if you want it :)

I'm still trying the above driver on my V2 & PAN but I've been unable to get telnet to accept. Probably hasn't installed WyzeHacks correctly. Still trying!

ryny24, 2021-02-08:
FYI - the cheap white $3 dongles off eBay from China do not work. Will confirm where to buy when I get one that works.

[root@v2_camera:ryan]# lsusb
Bus 001 Device 003: ID 0fe6:9700

gtxaspec, 2021-02-08:
this would work
https://www.amazon.com/USB-RJ45-LAN-Ethernet-Converter/dp/B0863YJB8W

ryny24, 2021-02-08:
Great! on it's way. Thank you.

ryny24, 2021-02-10:
My USB dongle arrived today, and you are correct. The V3 already has the drivers for it.  I created a custom script to UP eth0 and start dhcp. This works, I get an IP and can telnet into the camera over ethernet. Unfortunately, when Wifi is dropped, I also lose ethernet (ping & telnet). This is probably because the route table has the same subnet for wlan0 and eth0. Maybe wlan0 hasn't dropped yet, and the packets are still going out wlan0.

So I decided to keep Wifi off and reboot, it should bring up eth0 without wlan0 ever creating the duplicate subnet. Unfortunately, ping never starts for the ethernet IP, but I think this is due to the wait_wlan() function performed on startup. It's waiting for wifi, and thus, never runs my custom script.

I tried to modify wait_wlan() in main.sh to exit early without actually checking for wifi, but because the V3 has a read only file system, my changes are not saved on reboot.

CUSTOM_SCRIPT='/configs/eth0.sh'

/configs/eth0.sh 
ifconfig eth0 up
udhcpc -i eth0 -p /var/run/udhcpc_eth0.pid -b

ryny24, 2021-02-12:
I was able to get the camera to ping over ethernet on bootup. I had to modify a little in main.sh.   Telnet is still connection refused, so I guess WyzeHacks is not finishing. And the Camera doesn't come online until Wifi is connected. But, it's progress.

I have created a 'fork'. It's my first fork, but it looks like I did it right.
https://github.com/ryny24/WyzeHacks

thompatry, 2021-02-12:
Can you disable WiFi and force all the traffic to go through Ethernet? Just a thought to try. I am going to try to implement this on the v2.

ryny24, 2021-02-12:
How would you disable Wifi?   Here is where I think the problem is, the boot log shows the camera waiting at this spot:

[netService:1199]war: wpa_state=DISCONNECTED
 wiif disconnect
[0211 19:50:27.993][A][iCam][shellcall.c:548](no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[0211 19:50:28.004][A][iCam][shellcall.c:553](no.100000) msgque ret:[wpa_state=SCANNING
]

thompatry, 2021-02-12:
Could you put it in the script to do ifdown on the wlan?

ryny24, 2021-02-13:
I will try. I'm not sure it will satisfy the wpa_cli, but it's worth a shot. I'll try tomorrow.

MelonSmasher, 2021-02-19:
[The ethernet adapter that your using](https://www.amazon.com/dp/B0863YJB8W) uses the USB A port not the microUSB port? If so I'll purchase one to help, just want to make sure I get the right adapter. I bought [one of these](https://www.amazon.com/dp/B0876N3PMZ) hoping I could get power and ethernet working through just the micro USB port. No luck. I'm hoping you get it working and I can pair the setup with [one of these](https://www.amazon.com/dp/B01MDLUSE7).

gtxaspec, 2021-02-20:
use this with a v3:

https://www.amazon.com/gp/product/B07QGZK4FF

MelonSmasher, 2021-02-25:
So my parts came in today and I'm having some success on the V2 camera.

Here's my hardware:
- [Wyze Cam V2](https://www.amazon.com/gp/product/B07G2YR23M/ref=ox_sc_act_title_4?smid=ATVPDKIKX0DER&psc=1)
- [Axis USB type A to Ethernet](https://www.amazon.com/gp/product/B0863YJB8W/ref=ox_sc_act_title_1?smid=A1P0JKDBJQ43H7&psc=1)
- [UCTRONICS IEEE 802.3af Micro USB Active PoE Splitter](https://www.amazon.com/gp/product/B01MDLUSE7/ref=ox_sc_act_title_2?smid=A2IAB2RW3LLT8D&psc=1)

[image]

I'm attempting to get everything working manually before I try to script anything. Here's where I'm at:

[usbnet.ko](https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/driver/usbnet.ko) and [asix.ko](https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/driver/asix.ko) are on my NAS in the emulated SD card folder. I have them there for testing until I figure out a better place to put them. I was thinking about eventually splicing them into the official RTSP firmware.

I run the following:

```bash
insmod usbnet.ko
insmod asix.ko
ifconfig eth0 up
```

As expected I'm able to see the device in ifconfig:

```bash
eth0      Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:1038 (1.0 KiB)
```

However I run into a roadblock with DHCP. The device never gets a DHCP address. I get the following when I try to request a lease:

```bash
[root@WyzeCam]# udhcpc -i eth0 -p /var/run/udhcpc_eth0.pid -b
udhcpc (v1.22.1) started
Sending discover...
Sending discover...
Sending discover...
No lease, forking to background
```

If I manually assign an IP address I'm able to ping it from another machine and telnet into that address.

```bash
ifconfig eth0 192.168.0.28
[root@WyzeCam]# ifconfig
eth0      Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX
          inet addr:192.168.0.28  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

Any ideas what I'm doing wrong for DHCP to work?

MelonSmasher, 2021-02-26:
> Could you put it in the script to do ifdown on the wlan?

When I try that with a working ethernet connection everything works for a few minutes. After a minute or two the camera reboots. Seems like something is checking that the camera is connected to wireless and reboots the camera if it's not.

ryny24, 2021-02-27:
The boot log shows that Wyze is waiting for wpa_cli to return "COMPLETED".  Wyze does not finish booting until this. When wifi is not available, it wpa_cli will return "SCANNING, ASSOCIATING  or DISCONNECTED".

Unfortunately, we can't modify the wpa_cli since the file system is read only. very depressing.

`[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]              [exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING  
wpa_state=ASSOCIATING                                              
wpa_state=DISCONNECTED

[root@ETH-TEST-3:tmp]# printf '#!/bin/sh\necho "wpa_state=COMPLETED"\n' > /bin/wpa_cli 
-sh: can't create /bin/wpa_cli: Read-only file system`

I'll keep trying.

MelonSmasher, 2021-02-27:
@ryny24 I've been messing around with making a custom firmware based on [this video](https://www.youtube.com/watch?v=hV8W4o-Mu2o). I've been able to modify files on the firmware and flash it and then apply WyzeHacks on top with success. wpa_cli is on the JFS2 block within the firmware. I've been able to unpack the JFS2 system using [sviehb/jefferson](https://github.com/sviehb/jefferson) however I'm not sure how to repack the JFS2 system after making changes. It's been fairly trivial to make changes to the squashfs systems.

My thinkings is we may need to modify the stock firmware using the methods described in that video. Flash the modded firmware, then apply WyzeHacks on top. The mods to the firmware would include the kernel modules for USB ethernet and automatic detection of the USB ethernet adapter to choose Wifi or Ethernet.

If anyone has information on how to repack a JFS2 image I can put something together tonight / tomorrow.

ryny24, 2021-02-28:
@MelonSmasher wow, he makes that look so simple lol.   The only problem is, keeping wpa_cli functional when ETHERNET is not enabled in the config. Maybe we can create wpa_cli a link to another file system that we can modify. I will give it a try.

ryny24, 2021-02-28:
Doing a binwalk on the Wyze firmware file shows the same as the video. Multiple volumes including Squashfs. However, the firmware.bin which WyzeHacks sends is only a tar file. I think it's only sending the hacked scripts needed and not the firmware itself. Good idea actually. But, that will make it a bit more complicated to fix wpa_cli.

```ryan@240remote:~/tmp-wyze-ryny24$ binwalk firmware.bin 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             POSIX tar archive (GNU)```

MelonSmasher, 2021-02-28:
I made a tool that unpacks and repacks the firmware. It mounts the JFFS2 image read/write so that changes can be made. I've only tried on the Cam V2 firmware. Here's a link if your interested: MelonSmasher/WyzeFirmwareTool

MelonSmasher, 2021-03-02:
@ryny24 I have been able to modify the JFFS2 file system with some success. What would you do to wpa_cli if you could modify it?

MelonSmasher, 2021-03-02:
While not ideal, I found that killing wpa_supplicant after the camera has been running for a while allows the camera to stay up and running with no wifi connection forcing all traffic over eth0. I generated some Wyze Cam v2 firmware images using [my WyzeFirmwareTool](https://github.com/MelonSmasher/WyzeFirmwareTool). I've noticed that when wpa_supplicant is dead without the hacks in this repo the camera stays up and running. However when the hacks are enabled the camera reboots within a minute or two of wpa_supplicant being killed off.

Below are 4 different firmware images I generated for the cam v2. Two are based off the latest firmware from Wyze and two are based off of the RTSP firmware from Wyze. Two of them will kill wpa_supplicant after boot to force the connection over eth0 and the other two leave wpa_supplicant alone but do enable the ethernet adapter.

If I can't get the camera to stop rebooting, I may just work in the NFS mods from this repo into my firmware generator tool. 

[demo_4.9.6.218_telnetd_usb-eth_1614653385.bin.zip](https://github.com/HclX/WyzeHacks/files/6065993/demo_4.9.6.218_telnetd_usb-eth_1614653385.bin.zip)
[demo_4.9.6.218_telnetd_usb-eth_no-wlan_1614653513.bin.zip](https://github.com/HclX/WyzeHacks/files/6065994/demo_4.9.6.218_telnetd_usb-eth_no-wlan_1614653513.bin.zip)
[demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_1614653307.bin.zip](https://github.com/HclX/WyzeHacks/files/6065995/demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_1614653307.bin.zip)
[demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614653280.bin.zip](https://github.com/HclX/WyzeHacks/files/6065996/demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614653280.bin.zip)

gtxaspec, 2021-03-02:
didn't @HclX add a subroutine to reboot if wifi is lost? I forgot, I may be wrong, I'll check the source.

MelonSmasher, 2021-03-02:
Something else to note, here is that even with both wifi and eth0 enabled, traffic seems to be going over the ethernet adapter. While probably not the best way to measure this, but when I open the Wyze and look at the camera the ethernet adapter goes from a steady light to flashing continuously. I'm wondering if it's even worth killing the wifi connection.

gtxaspec, 2021-03-02:
you're saying that the camera works even with no wifi to initially connect?  I think some folks want to use this in areas with no wifi

MelonSmasher, 2021-03-02:
I don't think it will work as is for those who have no wifi. I'll try to explain what I've observed thus better, but they all require an wifi connection in some capacity.

## Ethernet and Wireless enabled:
- Works with WyzeHacks.
- Camera seems to prefer the Ethernet adapter over Wireless.

## Ethernet only:
- Reboots every minute or two with WyzeHacks.
- Requires wireless during boot for Wyze software to initialize. If wifi is killed to early something never fully starts and the Wyze app shows the camera as offline, even though it has network via eth0.
- If `wpa_supplicant ` is killed after a minute of being booted the camera runs fine without WyzeHacks. After `wpa_supplicant ` is killed all connections go through eth0 and the Wyze app works.

gtxaspec, 2021-03-02:
"Requires wireless during boot for Wyze software to initialize. If wifi is killed to early something never fully starts and the Wyze app shows the camera as offline, even though it has network via eth0."  this is what @ryny24 is talking about.  theres a program that runs and checks the output of wpa_cli.  if it sees the right text output from the program, it proceeds. 

feed that "wpa_state=COMPLETED" and you don't need wifi.

MelonSmasher, 2021-03-02:
so in theory replacing wpa_cli with a shell script that just echos "wpa_state=COMPLETED" should do the trick.

gtxaspec, 2021-03-02:
yes, the programs that the wyze cams run that monitors various states and variables are pretty rudimentary and mostly monitor the console output of various system utilities.

MelonSmasher, 2021-03-02:
I replaced wpa_cli with:

```
#!/bin/sh

echo 'wpa_state=COMPLETED'
```

Unfortunately the Wyze app sees the camera as offline and the back light flashes between yellow and blue, it never seems to fully initalize.

gtxaspec, 2021-03-02:
we will have to analyze the logs, maybe there are some other strings that must be echo'd in sequence...

gtxaspec, 2021-03-02:
`[exec-assis:274]: msgSndId:0
[exec-assis:289]: msgRcvId:32769
[exec-assis:369]: (no.100000) cmd:[/system/bin/sysMonitor.sh &]
[exec-assis:406]: (no.100000) system ret:[0]
[exec-assis:369]: (no.100000) cmd:[cp /system/local/cacert.pem /tmp/cacert.pem]
[exec-assis:406]: (no.100000) system ret:[0]
[exec-assis:369]: (no.100000) cmd:[insmod /system/driver/sensor_gc2053_t31.ko data_interface=1]
[exec-assis:406]: (no.100000) system ret:[0]
[exec-assis:369]: (no.100000) cmd:[echo "# HUALAI DNS #" > /etc/resolv.conf]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[/sbin/ifconfig wlan0 down]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[ifconfig wlan0 hw ether 7C:78:B2:0F:DC:29]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[/system/bin/restart_wlan0.sh]
[exec-assis:386]: (no.100000) popen  ret:[up
[exec-assis:369]: (no.100000) cmd:[killall -9 wpa_supplicant]
[exec-assis:369]: (no.100001) cmd:[echo "CST+8" > /etc/TZ]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[killall -9 udhcpc]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:386]: (no.100001) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[killall -9 udhcpd]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[killall -9 hostapd]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[killall -9 telnetd]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[sh /system/bin/restart_wlan0.sh]
[exec-assis:369]: (no.100001) cmd:[cp /usr/www /tmp/ -fr ]
[exec-assis:386]: (no.100001) popen  ret:[]
[exec-assis:386]: (no.100000) popen  ret:[down
[exec-assis:369]: (no.100000) cmd:[ifconfig wlan0 0.0.0.0]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[echo "# HUALAI DNS #" > /etc/resolv.conf]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[wpa_supplicant -D nl80211 -i wlan0 -c /tmp/wpa_supplicant.conf -B]
[exec-assis:386]: (no.100000) popen  ret:[Successfully initialized wpa_supplicant
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis[0202 22:00:17.903][A][iCam][shellcall.c:553](no.100000) msgque ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep wpa_state]
[exec-assis:386]: (no.100000) popen  ret:[wpa_state=SCANNING
[... cut ...]
[exec-assis:369]: (no.100000) cmd:[udhcpc -i wlan0 -H WyzeCam -p /var/run/udhcpc.pid -b &]
[exec-assis:386]: (no.100000) popen  ret:[udhcpc (v1.22.1) started
[exec-assis:369]: (no.100000) cmd:[wpa_cli -p /var/run/wpa_supplicant -i wlan0 STATUS | grep ip_address]
[exec-assis:386]: (no.100000) popen  ret:[ip_address=10.10.0.63
[exec-assis:369]: (no.100000) cmd:[echo "nameserver 8.8.8.8" >> /etc/resolv.conf]
[exec-assis:386]: (no.100000) popen  ret:[]
[exec-assis:369]: (no.100000) cmd:[echo "nameserver 8.8.4.4" >> /etc/resolv.conf]`

MelonSmasher, 2021-03-02:
@gtxaspec Thanks for the logs. I got it working! I replaced the following files with shell scripts:

- /system/bin/wpa_cli --> [bogus wpa_cli](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wpa_cli)
- /system/bin/wpa_supplicant --> [bogus wpa_supplicant](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wpa_supplicant)
- /system/bin/restart_wlan0.sh --> [bogus restart_wlan0.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/restart_wlan0.sh)

There are also two scripts that are started from the init scripts in /system/init:
- [wlan0_kill.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wlan0_kill.sh) -- cleans up unused processes, seems like this is required to complete initialization
- [eth0_init.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/eth0_init.sh) -- fires up eth0

Below is a firmware image for the Wyze Cam v2 that will enable the USB ethernet adapter, disable wireless, and enables telnet. I haven't tried the hacks from this repo yet, but fingers are crossed that they work with it.

[demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614706517.bin.zip](https://github.com/HclX/WyzeHacks/files/6070648/demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614706517.bin.zip)

MelonSmasher, 2021-03-02:
I've installed the hack on top of the custom firmware. The camera is up and running and the hacks installed. However my NFS server is not mounting. I'll keep monitoring it for any other issues see if I can see whats going on with NFS. Other than that, things seems ok, my password changed, and the hostname was set correctly.

MelonSmasher, 2021-03-02:
I tried to manually run `/configs/wyze_hack.sh` from a telnet session and now I can't get the camera to boot :(. I might have bricked it some how? Anyway just posting that to warn others that the firmware I made and that script do not like each other.

gtxaspec, 2021-03-02:
just reflash the stock firmware via SD card and start over.  these cameras are hard to brick

MelonSmasher, 2021-03-02:
Yeah I tried that, after getting the blue light when holding the reset button with the firmware on the SD card the camera clicks once and sits on a solid orange light.

gtxaspec, 2021-03-02:
sd card inserted,  you are holding down setup, with usb unplugged, keep holding, then plug USB in, keep holding 10-15 seconds? that should be the bootloader sequence

MelonSmasher, 2021-03-02:
Yup, thats what I'm doing. It's been the process for flashing the modded firmware I made as well. Of course I'm doing it with the stock firmware this time, but it should be the same. I guess don't play with fire if you don't want to get burned lol. Not a big deal, I'll get another one, kinda the cost of doing business.

gtxaspec, 2021-03-02:
have you tried to factory reset the unit? not flashing the firmware, but triggering the unit to wipe configs...now I recall I had a similar situation as you with modded firmware.

MelonSmasher, 2021-03-02:
How do you do that?

gtxaspec, 2021-03-02:
https://wyzelabs.zendesk.com/hc/en-us/articles/360031484511-How-to-factory-reset-your-Wyze-Cam-v1-v2-or-Pan

may take a few tries, it always does for me.

MelonSmasher, 2021-03-02:
Unfortunately no luck with that either :(

MelonSmasher, 2021-03-02:
I just ordered a new camera, until then I'm not going to be able to move this along any further. The firmware itself did not brick the device and neither did applying WyzeHacks to it. It all went downhill when I ran `/configs/wyze_hack.sh` from a telnet session. I'll be happy to help from the sidelines to figure out why NFS does not work with the modified firmware, but I won't be able to test directly.

gtxaspec, 2021-03-02:
I will duplicate what you did, which firmware file did you do? if I brick
my unit, I can hook up to serial to see what's going on.

MelonSmasher, 2021-03-02:
Awesome, I was on this firmware here: https://github.com/HclX/WyzeHacks/files/6070648/demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614706517.bin.zip

Just a heads up, I've been uninstalling WyzeHacks before flashing, then re-installing. Not sure if thats required, but it seemed like the cleanest way. That firmware has wireless disabled so you'll need to connect it to ethernet with and asix USB ethernet adapter.

You can also generate the firmware from this tool I made: [MelonSmasher/WyzeFirmwareTool](https://github.com/MelonSmasher/WyzeFirmwareTool)

```bash
sudo ./wyzefwtool -r -t -d
```

HclX, 2021-03-03:
nice work guys. I think this is the spirit of open source and collaboration. I probably won't integrate this into the main release since it requires quite some changes but only for an edge case.  Anyway, nice work and i will keep this thread open so people can see it easily.

I do have one question unrelated: how did you get files uploaded and shown as this URL: https://github.com/HclX/WyzeHacks/files/6070648/demo_v2_rtsp_4.28.4.49_telnetd_usb-eth_no-wlan_1614706517.bin.zip

It makes it feel like the file is published by me and might cause some misleadings.

MelonSmasher, 2021-03-03:
@HclX I made it with this script I made: [MelonSmasher/WyzeFirmwareTool](https://github.com/MelonSmasher/WyzeFirmwareTool). I uploaded it here so the other folks working on this could give it a try, I guess I should make a repo to host the firmware images outside of this thread. I'll do that and we can edit the comments so we can edit the links to point to my repo. Sound good? Sorry for the confusion.

MelonSmasher, 2021-03-03:
@HclX while your here, the firmware has the following changes:

- /system/bin/wpa_cli --> [bogus wpa_cli](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wpa_cli)
- /system/bin/wpa_supplicant --> [bogus wpa_supplicant](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wpa_supplicant)
- /system/bin/restart_wlan0.sh --> [bogus restart_wlan0.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/restart_wlan0.sh)
- [wlan0_kill.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/wlan0_kill.sh) -- cleans up unused processes related to wireless, seems like this is required to complete initialization
  - Script is started at the end of `/system/init/[app_init_da.sh | app_init_xiao.sh | app_init.sh]` 
- [eth0_init.sh](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/disable_wlan/support/mods/scripts/eth0_init.sh) -- fires up eth0
  - Script is started at the end of `/system/init/[app_init_da.sh | app_init_xiao.sh | app_init.sh]`
- [/driver/asix.ko](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/master/support/mods/ko/asix.ko)
   - `insmod /driver/asix.ko` added to `/system/init/[app_init_da.sh | app_init_xiao.sh | app_init.sh]`
- [/driver/usbnet.ko](https://github.com/MelonSmasher/WyzeFirmwareTool/blob/master/support/mods/ko/usbnet.ko)
   - `insmod /driver/asix.ko` added to `/system/init/[app_init_da.sh | app_init_xiao.sh | app_init.sh]`

Any ideas why NFS from HclX/WyzeHacks does not work with a firmware with those changed? I can explain the changes in better detail if I didn't explain them clearly.

HclX, 2021-03-04:
It's ok. You can start a repo to host the file, or just leave it here. If
you do the later, I'd suggest you create a readme file explaining how it
works and make sure  users know the risks and zip it together with the
binary file and then upload to this thread. It will also help others on how
to use it.

As for wyzehacks, I don't see anything related to WiFi so it should work.
What's the symptoms when you say it's not working?

MelonSmasher, 2021-03-04:
I'm working on adding pre-built firmware to my repo where I'll explain the risks (that I've become too familiar with) of flashing custom firmware.

NFS does not mount. The `mnt` directory and `media` directories are empty. I'd offer to give you more info than that, but I managed to brick my camera when doing so stuff I shouldn't have. My new camera should be here tomorrow.

It'd be great if we could get the custom firmware working with your Hacks. I just ordered eight cameras for my house and I really need POE USB ethernet and NFS lol!

HclX, 2021-03-04:
I see that you modified app_init.sh so I'd start debugging from there:
wyzehacks load itself before app_init.sh, and runs a certain things then
load app_init.sh. it is possible something before running app_init.sh is
waiting for network connection which won't happen until app_init.sh is
launched.

MelonSmasher, 2021-03-04:
@HclX I've replaced the firmware links, not sure if you want to remove the link from your comment? It's probably best that you do since telnet with the default password is enabled in that link. I've generated firmwares for each Wyze release here: https://github.com/MelonSmasher/WyzeFirmwareTool/releases telnet is is disabled on all of them, HclX/WyzeHacks can be used to enable telnet safely with a unique password.

HclX, 2021-03-04:
@MelonSmasher thanks for taking care of that. I have deleted my comment containing the link so we should be good now.

MelonSmasher, 2021-03-04:
Would you mind pointing me to the entry point where where wyzehacks loads? I'm not familiar with how it gets up and running. I'm wondering if It's as simple as just ensuring that ethernet loads up before wyzehacks.

HclX, 2021-03-04:
Sure, it's loaded in the following sequence (assuming you are working on v2 cam):
/etc/init/rcS 
/system/app_init.sh ==> symlinked to /configs/wyze_hack.sh
wyze_hack.sh extract itself into /tmp/wyzehack/run (forgot exact path, but something like that)
/tmp/wyzehack/run/main.sh
main.sh will check the md5 hash of /system/app_init_old.sh ==> this is the original app_init.sh
depending on the md5 hash, 
if there is a directory /run/wyzehack/run/init/<md5_hash>:
  it loads /run/wyzehack/run/init/<md5_hash>/init.sh
  init.sh calls /run/wyzehack/run/init/<md5_hash>/app_init_mod.sh
otherwise
  it loads /run/wyzehack/run/init/unknown/init.sh
  init.sh calls /configs/app_init_old.sh with some environment variables.

the md5 hash thing is because later wyze build added some nasty things in app_init.sh so we can't hook things non-invasively so we have to create a copy of app_init.sh and manually modify it, and the modification has to match the app_init.sh. 

So if your app_init.sh is not in the known list, we try our best by simply calling the original app_init.sh but NFS will not work.

[51 more replies]