HclX/WyzeHacks #63: SMB support?

pstoebenau, 2020-12-15:
I'm eager to try out this hack but unfortunately, my router only supports SMB. Is there a way to get SMB support? Can I install a package or something? Thank you for your work.

gtxaspec, 2020-12-16:
SMB may be too heavy.  Too much software and processing power required for smb
Which router do you have?

pstoebenau, 2020-12-16:
Yeah makes sense given the limited space on the cam. I have a netgear nighthawk rax38 router. It has a thing called readyshare which supports SMB, HTTP, and FTP. I would assume this would not be sufficient. Thanks for your help.

endertable, 2020-12-17:
Don’t give up pstoebenau.  This is one of the reasons the author created this software is for folks to create custom-ish firmware.  Space is usually not an issue because you can use your SD card to store extra library’s, scripts, and binaries as needed. I find the camera firmware usually does not mess with anything on the SD card except under the /record directory and maybe a few other directories.  There are lightweight binaries for FTP, ssh, and curl out there.  I haven’t come across any for SMB, but then I haven’t really been looking, so who knows.  Given the flexibility of your router with the other protocols, you could probably script to do what you need using something other than SMB.  Good luck, but continue to ask questions if needed.

pstoebenau, 2020-12-19:
Thank you. I might look into it. I initially wanted to make a hls stream in order to have a better quality stream than the rstp firmware Wyze has.

endertable, 2020-12-19:
Those are always welcome mods.  You have perked my interest to your ideas.

famewolf, 2021-01-03:
Instructions here tell you how to setup ftp which is in the list you said you could use.   https://github.com/HclX/WyzeHacks/issues/64#issuecomment-753556320

HclX, 2021-01-21:
I understand the need of SMB however, that's beyond the scope of this hack. I'm closing this issue now.

closed 2021-01-21 by HclX