HclX/WyzeHacks #64: Enable telnet without using NFS?

appleguru, 2020-12-16:
Is there some way to enable telnet (and maybe FTP or SSH/SFTP?) without enabling NFS support? I want to leave my recordings on the local microsd card, but access them remotely when desired.

appleguru, 2020-12-16:
I guess a related question: Are there any instructions for how to build a release image from source? I'd like to try some modifications/run my own script.

endertable, 2020-12-17:
I had the same request as I did not need NFS but wanted telnet.  Here are some steps

## To enable telnet go to:

https://github.com/HclX/WyzeUpdater/
grab  wyze_updater.py

## Click into "firmwares" directory
## grab  camera_telnet.bin

## Run the following command and get the MAC address of the camera you want to update:

./wyze_updater.py --user {WYZE_email} --password ${WYZE_password} list

## Then run the following, substituting '2CAA8Exxxx' with the MAC address of the cam:

./wyze_updater.py update -p 18080 -d 2CAA8Exxxx -f $PWD/camera_telnet.bin

## Confirm the camera with a 'Y' then you should see:

{Camera_IP} - - [{Date_Stamp}] "GET /firmware.bin HTTP/1.1" 200 -

## Within seconds, you can then telnet using root without a password.

## To make persistent across reboots, I updated /system/bin/app_init.sh and towards the top added:

killall -9 telnetd
busybox telnetd &

## The above 2 lines will prevent most iCamera binaries from killing telnetd

## This should get you started on building your cam empire.  I mainly used the built in wget to initially grab most of my stuff off my server but you can also put your scripts and other binaries on the SD card and access from either /media/mmc or /media/mmcblk0p1 depending on the firmware version.

appleguru, 2020-12-17:
/system/bin/app_init.sh doesn't exist on my camera; should I make it?

appleguru, 2020-12-17:
It looks like on my camera, /etc/init.d/rcS calls /system/init/app_init.sh

```
# Run init script
if [ -f /system/init/app_init.sh ]; then
    /system/init/app_init.sh &
fi
```
So maybe I need to edit that one?

appleguru, 2020-12-17:
Yeah, that seemed to work, but the root password got reset when I rebooted. Is there a good way to set it to something useful?

appleguru, 2020-12-18:
Ok, got it sorted; had to mimic the bind mount the telnet script was using...

Have a custom init script that looks something like this now:

```
#!/bin/sh

echo "Starting telnet..." > /tmp/custom_init.log              
killall -9 telnetd                                           
busybox telnetd &                                            
echo "Sleeping 31s so SD can mount..." >> /tmp/custom_init.log
sleep 31                                                      
echo "Setting root password..." >> /tmp/custom_init.log
umount /etc                                                                  
rm -rf /tmp/etc                                                              
cp -r /etc /tmp/                                                             
echo 'root:$1$MYSALT$1Sy1OLRk2kTa7P6fvzwp71:10933:0:99999:7:::' > /tmp/etc/shadow
mount -o bind /tmp/etc /etc                                                  
echo "Starting bftpd..." >> /tmp/custom_init.log                   
/media/mmc/bin/bftpd -d -c /media/mmc/bin/bftpd.conf               
echo "Custom init complete..." >> /tmp/custom_init.log
```

That I run from `/system/init/app_init.sh` via a line at the start like this `/system/init/custom_init.sh &`

I copied bftpd from https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/tree/master/firmware_mod/bin and that worked surprisingly well on the stock fw!

endertable, 2020-12-18:
I'm sorry, I gave you the wrong path for app_init.sh.   You were right on where yours is.  I don't know if it will accept that password hash, that looks like md5 I think you need the good old fashioned DES crypt which will only be like 13 characters.

appleguru, 2020-12-18:
It works great (at least on this wizen camera/fw); generated the hash like this:

```
#   openssl passwd -1 -salt <YOUR SALT> <YOUR PASSWORD>
```

per https://github.com/HclX/WyzeHacks/blob/master/installer/config.inc.TEMPLATE

endertable, 2020-12-18:
Sounds good.  Looks like your getting up and running..

famewolf, 2021-01-02:
Any tips on removing nfs and reverting to using sdcard if you did the full install?  How do I pick the salt to use with my choice of password?

endertable, 2021-01-02:
I think what I would do is reinstall your WYZE current firmware which will disable the NFS, then follow my instructions above to just enable telnet.  Use any alphanumeric characters for your salt.  If there’s a specific reason you don’t want to have to reinstall the firmware then come back here and state that and I can give you a step-by-step to do so

famewolf, 2021-01-03:
Doing a reset isn't as easy as it sounds.  I'm disabled and some of my cam's are mounted high on the house.  I had other people install them.  When it appeared to work on the test cam I then pushed it to all the cams...now I've seen it's unreliable..if the nfs share goes down it quits saving until you go into the wyze app and tell them all to reboot 1 at a time.  As an alternative if someone can tell me how to have it check and remount every x minutes that would be a viable alternative.  I just can't have my cam's offline for 12 hours because the pi got rebooted in the house and I didn't manually reboot all the cams.

If he has a remote install, he ought to have a remote uninstall.  I can telnet into the camera.  I know I could delete his fake mount command but not sure what else had to change to stop it from mounting the nfs share.  Ideally I'd like to have it store local but use some form of rsync like file transfer to keep a 2nd copy in sync inside house.

adamurban, 2021-01-03:
Take a look at your /system/init/app_init.sh

You should be able to restore it to factory and/or comment out the lines that launch the NFS hooks.

famewolf, 2021-01-03:
app_init.sh is symbolically linked to /params/wyze_hack.sh which when cat'd appears to be a binary file.

appleguru, 2021-01-03:
Does /system/init/app_init_orig.sh exist?
 (If so, remove the symlink and move that file back to /system/init/app_init.sh )

famewolf, 2021-01-03:
It does exist and is a shell script but I don't want to lose telnet.  Replacing it puts me 100% back doesn't it?

appleguru, 2021-01-03:
You can add 

```
killall -9 telnetd                                           
busybox telnetd &
```

to the script to keep telnet.

if you do lose it, it's easy to restore using the steps @endertable posted above (wyze_updater.py with camera_telnet.bin).

appleguru, 2021-01-03:
Also, for anyone else that comes across this thread, here's a short summary of the above on how to enable telnet, FTP, and set your own root password without losing any factory functionality.

(I did this all on my Mac, but any *nix like system should work)

1. Download https://github.com/HclX/WyzeUpdater/
2. Authenticate wyze_updater.py and enumerate your camera by running: `./wyze_updater.py --user {WYZE_email} --password ${WYZE_password} list`
3. Run `./wyze_updater.py update -p 18080 -d YOUR_SN_FROM_STEP_2 -f firmwares/camera_telnet.bin` After a few seconds you can ctrl-c to exit. (This step uses Wyze’s update mechanism to run a small script that blanks out the root password and starts telnet so we can access the system)
4. `telnet ip.address.of.camera` (login as root with no password)
5. Use vi on your camera to add `/system/init/custom_init.sh &` to the beginning of the existing init script: `vi /system/init/app_init.sh`
6. On your computer (not on the camera, it doesn’t have openssl), generate a password hash to store on your camera so you can login with your own password in the future: `openssl passwd -1 -salt <YOUR SALT> <YOUR PASSWORD>`
7. Create your custom init script: `vi /system/init/custom_init.sh` You can paste in this as a starting point, swap in the output from step 6 for the placeholder below:
```
#!/bin/sh

echo "Starting telnet..." > /tmp/custom_init.log              
killall -9 telnetd                                           
busybox telnetd &                                            
echo "Sleeping 31s so SD can mount..." >> /tmp/custom_init.log
sleep 31                                                      
echo "Setting root password..." >> /tmp/custom_init.log
umount /etc                                                                  
rm -rf /tmp/etc                                                              
cp -r /etc /tmp/                                                             
echo 'root:$1$MYSALT$1Sy1OLRk2kTa7P6fvzwp71:10933:0:99999:7:::' > /tmp/etc/shadow
mount -o bind /tmp/etc /etc                                                  
echo "Starting bftpd..." >> /tmp/custom_init.log                   
/media/mmc/bin/bftpd -d -c /media/mmc/bin/bftpd.conf               
echo "Custom init complete..." >> /tmp/custom_init.log
```

8: `chmod +x /system/init/custom_init.sh`
9. (someone should just host this on a non-https site so we can just wget it directly from the internet…) Download bftpd from here and copy the file over to your SD card at `/media/mmc/bin/bftpd` : https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/bin/bftpd
10. Download the bftpd config file from here and copy the file over to your SD card at `/media/mmc/bin/bftpd.conf` : https://github.com/EliasKotlyar/Xiaomi-Dafang-Hacks/blob/master/firmware_mod/config/bftpd.conf
(Note, for steps 9 and 10 I just served the files locally with python and downloaded them on the camera via wget, `python -m http.server 8000`
11. `chmod +x /media/mmc/bin/bftpd`
12. `reboot`
13. 
That’s it, now every time the camera starts, it will enable telnet, set the password you specified for the root password, and start bftpd. You can now access the cameras filesystem (and SD card) with any FTP client!

famewolf, 2021-01-03:
In the directory /tmp/run/wyze_hack  the file mount_nfs.sh has the following code in it:

if [ -z "$NFS_ROOT" ];
then
    echo "NFS_ROOT not configured, skipping NFS mount..."
    exit 1
fi

which looks like if you comment out the NFS_ROOT export line in config.inc you would skip the nfs mount while still retaining other features like the auto reboot.   I may play with this before trying the method suggested because I don't think just replacing the init is going to be enought to stop everything it's doing.

famewolf, 2021-01-03:
So here's a summary of my findings....commenting out nfs_root doesn't appear to do the job.   restoring the original init and adding the 2 lines to start telnet works in that telnet does indeed start however the password is neither blank nor the password I WAS using.  Repushing the code to the camera gets me back to where I started.  

While the instructions appleguru summarized may work for someone starting from a clean "reset" I'm not sure they are going to work as well for someone already running the wyzehack code.

endertable, 2021-01-03:
Restoring /system/init/app_init_orig.sh back to /system/init/app_init.sh then adding the 2 telnet lines close to the beginning of the script (not using custom script) will disable NFS and keep the default “ismart12” password.  It should kick in the SD card recordings.

adamurban, 2021-01-03:
> So here's a summary of my findings....commenting out nfs_root doesn't appear to do the job. restoring the original init and adding the 2 lines to start telnet works in that telnet does indeed start [...]

I had the same experience RE the default root password not working on the latest firmwares with just telnet enabled, so set my own. If you do the same you should be good to go.

famewolf, 2021-01-03:
Thanks for all the replies.  Apologies for "hijacking" the thread.  I think appleguru should change the title or create a new issue with "How to add telnet and ftpd while keeping sd recordings" and put those instructions there (or link to them here) so people hunting for similar methods can follow it.  I linked to appleguru's instructions in an issue requesting samba support because ftp was one of the methods he said he could use (with his router).   I'll give this another go once I have some caffeine!

HclX, 2021-01-21:
back to the original topic: telnet is now fully enabled without the need of NFS, all you need is simply comment out the NFS_ROOT settings in configuration file. Be warned many features depend on NFS shares, so by commenting out NFS_ROOT, most of them are disabled, the only ones left are telnetd, custom password shadow, and auto reboot.

HclX, 2021-01-21:
this is on `dev` branch, release 0.5.01.

famewolf, 2021-01-21:
> back to the original topic: telnet is now fully enabled without the need of NFS, all you need is simply comment out the NFS_ROOT settings in configuration file. Be warned many features depend on NFS [...]

Any chance of offering ftpd support to camera for when not using nfs?  It's documented above with a binary already available.

HclX, 2021-01-21:
@famewolf, the reason of not offering ftpd or smb or similar things is the extra binary needs to be stored somewhere: without NFS the only way to get the binary onto the camera is to package it inside of wyzehack binary. The hack binary is stored in /configs partition, which has only 128KB freespace. Including any binaries will significantly increase the file size and cause this partition to be filled up, to make it worse, this partition can not be cleaned by an SD card recovery so once it's filled up the camera might be permanently bricked.

If you really need those, here would be my solution:
With latest version wyzehack will be able to use physical SD card when NFS_ROOT is unset. You can put any files onto an sdcard, and then use CUSTOM_SCRIPT feature to run a script from sdcard to load any binaries you want. You will need to set your CUSTOM_SCRIPT to "/media/mmc" to get it work.

HclX, 2021-01-26:
marking this as fixed.

closed 2021-01-26 by HclX