HclX/WyzeHacks #66: How do I uninstall so it uses sdcard again without reset?

famewolf, 2021-01-01:
I installed the hacks however they don't appear to be recording to my nfs share like they should be....how can I revert at least enough of the changes so it uses the sdcard again?  I'd like to avoid having to hard reset them as some are mounted 10-15 feet up on an outside wall.

HclX, 2021-01-12:
Currently a factory reset AND a sd card firmware recovery is the recommended way to completely remove this back. If you are familiar with linux system, you can manually telnet onto the device and revert the changes to app_init.sh.  I will see if I can make some changes to make uninstall easier.

famewolf, 2021-01-12:
I use only linux so I'd say I'm familiar from an average user standpoint.   I attempted removing the symbolic link for app_init.sh and restoring the original but that didn' seem to restore it...then again I was trying to retain telnet and ftpd while just removing the nfs portion and as I recall the issue I hit was it lost the custom password I was using and at the time I didn't know the one YOU were using.

HclX, 2021-01-21:
check the latest release 0.5.01 on `dev` branch: you can now tell it to uninstall itself by placing a special file in the NFS camera share.

famewolf, 2021-01-25:
I have it installed and even looked at the code for that release and I STILL can't find where you are checking for a file to uninstall.

HclX, 2021-01-25:
Implementation is here:

https://github.com/HclX/WyzeHacks/blob/a0fa1425318aef8d39a627b3b69c2ccb3e683d40/wyze_hack/main.sh#L279

Keep in mind this is in dev and dev_v3 branch only. I haven't integrated to master branch yet.

famewolf, 2021-01-25:
Thank you.   I think I was only looking at a single commit and not all your changes.    I've currently moved the nfs partition to a different raspberry pi and am still giving it a go.  I was thinking about  using a script to create daily mp4's from all the little files.  I borrowed the code from a forum post and adapted it to the layout of the nfs share.  I've attached it in the event you might find it useful.

cd /mnt/nfsshare/WyzeCams/
for device in $(ls); do

cd /mnt/nfsshare/WyzeCams/$device/record
for day in $(ls); do
#only executes loop for day folder
ls $day//.mp4 || continue
#skip folders already concat
[ -f $day.mkv ] && continue
#create txt lists
echo “Compiling $day for $device…”
ls $day//.mp4 > $day.lst
#Prepend txt lists
sed -i -e ‘s/^/file /’ $day.lst
#Concat each day
ffmpeg -nostats -f concat -i $day.lst -c copy $day.mkv
done
done

echo “done”

closed 2021-01-26 by HclX