HclX/WyzeHacks #67: Validate NFS is not working?

isaacrlevin, 2021-01-01:
Just went through the steps yesterday to set this up on a Wyze cam, got notification that the firmware was updated by the camera. I do not however see files being put on to my NFS share, and there is no SD card detected by the Wyze app . Is there a way for me to determine if my NFS settings are incorrect , as I am able to mount the share onto a Raspberry Pi

famewolf, 2021-01-02:
Try reinstalling several times/rebooting the camera's.  Took a while to get my 4 writing to the nfs...worse...if the nfs ever goes down even briefly (ie the pi having to reboot) some of the camera's seem to go into limbo and never recover....they just quit writing entirely til the cam gets rebooted.

isaacrlevin, 2021-01-02:
@famewolf does it work better with or without an SD card installed while running the script.

famewolf, 2021-01-02:
I'd say it runs the same...the only reason I have sd cards in is because they were ALREADY in and mounted high on outside of house so not easily removed.  I like the idea of it writing to nfs but personally I'd be just as happy if it wrote to BOTH places so I could have it in the camera for when wifi drops/power fails (you can put battery packs on the cam wires to keep the camera's going for HOURS) while also giving me a central point where the video is saved like more expensive camera systems.  I'm using a raspberry pi with a hard drive dock plugged into the usb 3.0 port which allows for rotating drives on a monthly basis for example. (So if for some reason you need to see last month's footage you'd have it availabe).

HclX, 2021-01-18:
@isaacrlevin  telnet into the camera and try to mount it manually. there are many settings on an NFS server and the kernel of the camera was built probably only for a specific combination of settings. If you can manually mount the share from the camera, the same settings should work.

closed 2021-02-17 by HclX