HclX/WyzeHacks #69: Feature Request: Support the Wyze Doorbell Cam which has NO microsd at all.

famewolf, 2021-01-04:
One of the biggest disappointments for many people on the wyze doorbell cam was that it doesn't have a microsd for local storage.  If we can mount a nfs share identical to how we are doing now it can be used along with wyzecam-hls and shinobi.

gtxaspec, 2021-01-06:
interesting.  it uses the same SoC as the wyzev v3.

famewolf, 2021-01-06:
I don't think it's even shipped yet but I have one pre-ordered.  We'd also have to come up with some method to review the footage...I saw a scripts which will take an sdcard and create a mp4 of an entire day for viewing....this would be useful for all the cams as wyzecam-hls's thread is unreliable...I could only get it to work with shinobi but even there it's missing alot of time that IS being saved to the nfs share because I can view it from inside the app.

HclX, 2021-01-09:
I have one and confirm they are very similar to the camera except the lack of sd card. I haven't really looked at it yet, it's on my list but busy due to holidays:)

cheme75, 2021-01-26:
I’m beta testing so have several permutations of the hardware from orig to prerelease. I only opened up one due to an assembly glitch that pinched and split the WiFi antenna wire which I reported and got a replacement. Others told me the cam itself was same as v2. However, they did a hardware revision and I only have the one so haven’t opened it, appears to be the pre-release retail version. The cam may be more like outdoor cam, which I think is same as v2, don’t think it’s the v3 cam since it doesn’t seem to have same low light ability. I’m only a chem engr, not into coding, but I’d love to try loading this fw onto an older unit to make it useful as a monitoring cam. So far the old units still work but with no ssd card and no cloud connection, they are simply dumb cams for live streaming, plus they could block them in the retail app making them totally useless.

famewolf, 2021-02-02:
I just received my pre-order video doorbell today and am reviewing the install instructions on youtube.   I should be available to beta test a version should you need another tester.

gtxaspec, 2021-02-02:
there is a hidden micro USB port above the power contacts under a sticker, the unit can also be powered using micro USB.  just fyi

famewolf, 2021-02-02:
> there is a hidden micro USB port above the power contacts under a sticker, the unit can also be powered using micro USB. just fyi

Have you checked if an OTG cable with a sd card reader would give it a sdcard?  Some of the OTG cables are tiny and so are the readers.  Thanks for the info!

gtxaspec, 2021-02-02:
no clue, I discovered it from reviewing the fcc docs, and it was confirmed by another user on the wyzecam subreddit.  still waiting for them to come back in stock so I can get a few.

famewolf, 2021-02-02:
> no clue, I discovered it from reviewing the fcc docs, and it was confirmed by another user on the wyzecam subreddit. still waiting for them to come back in stock so I can get a few.

Just to confirm...you are not talking about the sticker that has the qr code?   I have what appears to be a rubber plug above the terminals as opposed to a sticker.

gtxaspec, 2021-02-02:
another user said it was a sticker, it could be a rubber plug.  ref https://fccid.io/2AUIUWVDB1/Internal-Photos/Internal-Photos-4957362.pdf

cheme75, 2021-02-02:
It’s not a plug, just a rubber sticker to keep moisture out of the USB port. It peels off once you can lift an edge to grab it. I assume it’s how they load up the initial firmware. I’m powering a couple of early beta models just for viewing the live stream since they still connect to the beta app. But haven’t been able to connect one to my pc yet.

famewolf, 2021-02-02:
> It’s not a plug, just a rubber sticker to keep moisture out of the USB port. It peels off once you can lift an edge to grab it. I assume it’s how they load up the initial firmware. I’m powering a [...]

Thanks...I wasn't being enthusiastic and pulling it up till I made sure I didn't mess something up.  The tears in picture are huge compared to actual viewing.

famewolf, 2021-02-02:
Something like this:   https://www.amazon.com/Ribbon-Degree-Angled-Female-Charging/dp/B07Q9K64PN/ plus a microsd card reader like the one I listed above might work.  Hopefully there are cheaper versions on ebay.

cheme75, 2021-02-02:
I’m really new at this so will have to read on how to hack the V2 and see if maybe there’s some insight on getting into the db.

famewolf, 2021-02-02:
> I’m really new at this so will have to read on how to hack the V2 and see if maybe there’s some insight on getting into the db.

Well HclX is currently working on the v3 cams so might be a while although the "send a modded firmware update" method seems to be a "universal" solution for now since the updates are unsigned.

HclX, 2021-02-02:
Guys, I took a brief look at the firmware of the doorbell. I don't think
recording onto nfs share is possible as they don't have that code compiled.
It's possible to enable telnet support (which I remember I already enabled
in latest release) but not much you can get from there.

Also, the micro usb port might be a usb otg port but I don't have a cable
splitter to verify that.

thompatry, 2021-02-02:
Could you possibly take the compiled NFS execs and libs from the Cam v3 and put them on the doorbell?

HclX, 2021-02-02:
That might be possible, but you will get a v3 camera and lose the doorbell
functionality:)

famewolf, 2021-02-02:
Could the doorbell ftp to a ip on the lan (with or without user/password) and upload the 1 minute files..perhaps maintaining the connection?

HclX, 2021-02-02:
No need to use ftp, I can get NFS mounted. The problem is camera won't
record anything on filesystem, so no way to get the video out without
drastically change the camera executable.

gtxaspec, 2021-02-02:
I see, so the doorbell does not record ANY video to the file system, i wonder how and where it caches it and uploads it to the cloud.
And there still isn't a way to make the video devices accessible by multiple programs I assume...

famewolf, 2021-02-02:
Well crap...I was hoping it at least kept the previous minute until the current one overwrote it.

thompatry, 2021-02-02:
You would have to looking into the connecting to the camera node. I know someone out there was working on something like this for the latest Cam v2 firmware. The were trying to get RSTP to work while keeping the latest firmware.

cheme75, 2021-02-02:
The db just does the default 12sec but it’s camplus capable to there must be a tweak to disable the 12 sec limit.

cheme75, 2021-02-02:
It does seem to upload and replay from the cloud since a lot of replay problems have been due to the servers. Rtsp fw would be nice to see happen.

famewolf, 2021-02-02:
> It does seem to upload and replay from the cloud since a lot of replay problems have been due to the servers. Rtsp fw would be nice to see happen.

Bad design on wyze's part...not all of use have high speed internet  (I use a wifi hotspot and get 10 to 16 megabit...funny how that's slow nowadays but a while back that was the best you could get from cable tv).  Not being able to view a cam locally is already a major pain since it uploads it all so wyze's server can download it to my phone.  If I view to many at once it starts telling me to downgrade quality.

cheme75, 2021-02-02:
Ya, I’m not thrill with the wyze. I bought a V2 simply to watch front door area for package delivery notifications since they advertised free person detection. Now it’s a subscription. I bought a set of plugs and they work ok but they just replace old timers that also worked fine, just more convenient for operating via app or by voice with Alexa. Got a pair of bulbs free in a promo deal and they are ok. I don’t really have much need for automation beyond simple toggling on at sunset and off at bedtime. The big weakness became very apparent when our internet service junction box got flooded and service started dropping out. None of the wyze stuff worked. The dbell actually failed completely refusing to reconnect. They sent a replacement which has been ok and I returned the failed unit but I never heard about the cause. Got a new line run from junction box to house so at least internet had been ok again.

famewolf, 2021-02-02:
Well I gambled with my pre-order and guess I just barely got my $50 worth unlike the thermostat which was considerably more bang for the buck.  I mean I already have a wireless doorbell that rings both the normal chime plus a plug in one on opposite side of the house and could have easily mounted a v2 cam where it could record folks approaching the door.   I hesitated when I was writing to microsd because a camera that low could literally be ripped off the wall and taken with whoever might be up to no good but now I'd at least have the nfs feed.  Win some and lose some I guess.  I need to setup my raspberry pi to upload the video files to a cloud service.  There is a script called wyzecombiner someone wrote that will takes all the little files and spit out a daily video file.

cheme75, 2021-02-02:
We live in so flo due to wife’s job relo to Miami so did a lot of research. Luckily I had a cousin who lived here a while and my wife had good connections at work so we’re advised of best locations to settle. We chose south west broward county rather than Miami and it’s been a good decision from low crime to one of the lowest rates of covid in all of so flo. Anyway, we’re in 24/7 guard gated community with just one access road so exceptionally low incidents of any sort here. I was surprised to read on nextdoor a person a few blocks over had pkgs stolen. Probably a one off since we never heard any fallout or repeat offenses. I looked at the available db cams but most all were expensive and required subscriptions so turned me off. Looked at alt cams and saw quite a few, wyze had good reviews and free person detection with 14 days cloud stg.  So for $20 I figured I’d setup a cam to watch front, we have a sidelight window next to front door and a privacy film so no one can see in. Original person AI was ok, missed at times since I have to keep cam on it’s side to get view of both the landing for packages and vertically for faces, and being inside wasn’t able to use IR lights on it at night. But there’s an overhead sensor light that works well to illuminate the area. Long story, it works for our needs. The new wyze AI is better finally, and it’s grandfathered into a pay what you want for the person detection, I pay zero since that’s why I bought it. I’m following these hacks since we plan to relo out of so flo for retirement and I can see benefits of using these cams for security in more open communities, but I refuse to get sucked into a subscription. Id probably look to the rtsp firmware so. I could use better software and these hacks for local stg on my NAS.

famewolf, 2021-02-02:
Sadly the rtsp firmware is WAY out of date and you'd totally lose all the cloud upload. I've also read it's flaky. I've debated the issue myself as I have a couple of nas's as well and have played around with shinobi and a couple of other cam monitoring packages.

cheme75, 2021-02-02:
I thought I saw an updated rtsp fw not too long ago on the wyze forum, but ya, reading the forum, it does seem full of flaws - I've avoid it so far, but I keep it mind for future use. Right now If I had to pick a retail option, I'd probably go with eufy for low end security monitoring. I was thinking I'd take out the cam unit from a doorbell and see if I could connect it to a rasp pi somehow - it's a bit beyond my typical electronics skill - the most I've messed with was ages ago - soldering in a jumper to fix a mobo per mfg's instructions back in the 386 days.

famewolf, 2021-02-02:
I do not have a soldering "skill". I have a soldering "attempt" that may or may not work while looking UGLY. I've debated taking a cam apart...got one that works fine in day but at night is "cloudy" using the ir light...not sure why. Got a dead one on my desk here that has the yellow light on all the time and is unresponsive to reset or anything else. Figured I'd try it first.

cheme75, 2021-02-02:
HaHa! Been soldering since I was a kid - I built a fuzz pedal for a cheap electric guitar I had when I was a young teen. My dad built our home stereo from a kit so we had plenty of stuff around. I got into stained glass work in the 80s and got good at soldering, although it's not as fine a level as for electronics. Anyway, if you find some success, would be nice to hear the details.

famewolf, 2021-02-02:
> @famewolf
> 
> I think you may be misinforned. the rtsp firmware was just recently updated, within month or so, and to my knowledge included all features from the mainline firmware.

I would LOVE to be wrong.  All features including cloud storage or is that still a loss?   Stable now or still flaky?

yortnoswad, 2021-02-05:
I noticed nobody replied to this.
Here is the link:
https://forums.wyzecam.com/t/new-rtsp-binaries-for-beta-testing/145275
Where it says the following
------------------------

Hi All,

We have a new version of RTSP firmware for testing! To make sure we capture
all issues, please only use this forum thread to report RTSP issues.

Links:
v2:
https://download.wyzecam.com/firmware/rtsp/demo_v2_rtsp_4.28.4.49.bin.zip
726
<https://download.wyzecam.com/firmware/rtsp/demo_v2_rtsp_4.28.4.49.bin.zip>
Pan:
https://download.wyzecam.com/firmware/rtsp/demo_Pan_rtsp_4.29.4.49.bin.zip
248
<https://download.wyzecam.com/firmware/rtsp/demo_Pan_rtsp_4.29.4.49.bin.zip>

RTST installation/setup instructions:
https://wyzelabs.zendesk.com/hc/en-us/articles/360026245231-Wyze-Cam-RTSP
769
<https://wyzelabs.zendesk.com/hc/en-us/articles/360026245231-Wyze-Cam-RTSP>

Change list:

   - Added RTSP functionality based on 4.X.6.199 firmware
   - Applied security update since previous version of RTSP firmware

Note:
This is a one-time effort from Wyze to support RTSP functionality on Wyze
Cam v2 and Pan. We are not committing to ongoing maintenance for RTSP
firmware. New features will not be updated on the firmware.

Enjoy the new binaries!

Your Wyze Team

------------------------

famewolf, 2021-02-05:
@yortnoswad  I appreciate the info.....but you didn't answer the one question I asked....do you still lose all cloud storage?

yortnoswad, 2021-02-05:
Tried it after you just asked.  The answer is no, you do not loose cloud
storage

I setup a wyze cam with the above firmware, and setup the rtsp according to
the directions.
I connected a computer up to the rtsp feed and was streaming the feed.
I enabled motion dection, then created motion (walked in front of the
camera).
The rtsp feed continued to work, and it uploaded the motion to the normal
wyze cloud storage.

I did not test if it also did nfs writing at the same time.  I don't have
nfs writing setup.

famewolf, 2021-02-05:
> Tried it after you just asked. The answer is no, you do not loose cloud storage I setup a wyze cam with the above firmware, and setup the rtsp according to the directions. I connected a computer up [...]

Thanks...that's great news I'll pass on to some of my friends with cams.   I appreciate you going to that effort. I just figured you might know if you were using it.

adamurban, 2021-02-22:
Finally got my wyze doorbell installed. Any tricks to getting telnet going @HclX ? Or do the normal instructions “just work”?

appleguru, 2021-02-22:
Finally got my wyze doorbell installed. Any tricks to getting telnet going @HclX ? Or do the normal instructions “just work”?

When I tried the remote installer, it only shows 1/4 of my wyze cams. (doesn't list the v3 cams or the doorbell)

famewolf, 2021-02-22:
He's still working on v3 support so I would assume both are in the dev branch and not stable.

appleguru, 2021-02-22:
> He's still working on v3 support so I would assume both are in the dev branch and not stable.
> […](#)

Thanks, I did a stupid and was on my (out of date) local copy and not the latest master. Latest master sees v3 cams and the doorbell. dev_v3 branch sees v3 only. dev sees all. None seem to enable telnet for the doorbell though; will try manually with wyze updater next..

HclX, 2021-02-22:
@appleguru Please use wyze updater approach, it works and will not change anything permanently. Keep in mind, if you brick it, you can't recover it since there is no SD card slot.

appleguru, 2021-02-22:
@HclX I wonder if manually adding a `sdcard_exist=1` flag to the `[SETTING]` block in /etc/config/.user_config would make a difference..

HclX, 2021-02-22:
I haven't looked into this thoroughly, but a quick scan in the firmware
shows no code referring sd card at all. This is understandable as they
don't even have sd card slot available so it's reasonable to assume they
don't have code dealing with it. The best thing we can get is to copy the
video to nfs share manually if they ever record it somewhere before
uploading it. But I'd say the chances are slim.

adamurban, 2021-02-22:
Semi related question… what’s the easiest way to grab a file off the camera? And has wyze published firmware for this somewhere yet?

appleguru, 2021-02-22:
Semi related question… what’s the easiest way to grab a file off the camera? And has wyze published firmware for this somewhere yet?

HclX, 2021-02-22:
They don't publish the firmware binary since there is no need for users to
get it (because you can't do sdcard recovery). I got the first update image
by inspecting the https traffic of the app when there is an update
available. And I figured out how to run custom script on the device based
on that image. After that I got telnet enabled then you can simply btelnet
onto the device and copy file to a NFS share.

appleguru, 2021-02-22:
Thanks... one other config flag that looks (potentially) useful: `storage_device=1`

appleguru, 2021-02-22:
And finally `/backupa/localPic.jpg` looks interesting

(mostly leaving these notes here for myself to pickup once I have NFS or another way of grabbing files up an running)

famewolf, 2021-02-22:
> Semi related question… what’s the easiest way to grab a file off the camera? And has wyze published firmware for this somewhere yet?

In one of the threads about telnet one of the users posted instructions on how to install ftpd on the camera so you could log in and get the files.....this might be an alternative if nfs is a no go.

appleguru, 2021-02-22:
> In one of the threads about telnet one of the users posted instructions on how to install ftpd on the camera so you could log in and get the files.....this might be an alternative if nfs is a no go.

Oh, duh... bftpd. That was my post; transferred the binary and config over to /tmp via wget and ran; works great: `./bftpd -d -c /tmp/bftpd.conf`

`/backupa/localPic.jpg` is a photo of me from earlier today from when I was messing with the camera. I wonder what triggers that? (Tamper sensor maybe??)

gtxaspec, 2021-02-22:
have you tried to set storage_device=1 ?  you can run the command 'logcat' to get some icamera logs too.  not sure if there are duplicated in the normal boot.log

appleguru, 2021-02-22:
1 appears to be the current default for the doorbell. I have not tried changing it yet.

```[root@hualai:mnt]# cat /etc/config/.user_config
[SETTING]
firstBindFlag=0
osdLogoSwitch=2
faceSwtich=1
white_face_switch=1
white_brightness=100
white_switch=2
storage_device=1
alarmInterval=180
alarmDate=255
pir_alaram=0
speaker_vol=50
watermark_flag=0
low_bitrate=30
motor_limit=0
limit_min_y=0
limit_max_y=0
limit_min_x=0
limit_max_x=0
exposureModel=0
ubootflag=0
preset_y=0
preset_x=0
slide_y=0
slide_x=0
motionback=1
AALY=50
AALX=50
AASY=25
AASX=25
motionCruisin=2
motionTrack=2
AST=1
MAT=0
AMALevel=100
MMALevel=150
timezone=-5
drawBoxSwitch=1
TBDuration=1440
TBStart=0
CASwitch=2
SASwitch=2
AALevel=5
AASwitch=2
MALevel=10
MASwitch=2
recordType=1
telntSwitch=2
logUdisk=1
logSd=1
osdSwitch=1
verSwitch=1
horSwitch=1
fps=20
res=1
bitRate=180
nightVision=3
indicator=1

[NET]
p2pid=
password=
sub1gMac=
r_encr=
ssid=
[root@hualai:mnt]# 
[root@hualai:mnt]# iCamera
[02/22/2021 10:49:02]===========welcome to iCamera_app===========
[exec] msgSndId:0
[exec] msgRcvId:32769
msg productservices/communicate/msg/msg.c,77:[msg]dbg: msg_queue_create() ok  MsgId:65538
[timer]dbg: init complete.
[02/22/2021 10:49:02]watchdog_init() ok  TimerId:1
[02/22/2021 10:49:02]Metric server start
[02/22/2021 10:49:02]Metric msgQueueId create 98307
[02/22/2021 10:49:02]Metric msgQueueId init finish 
[02/22/2021 10:49:02]Create worker thread
[02/22/2021 10:49:02]pthread_create ret = 0
[02/22/2021 10:49:02]<Reading> /params/config/.product_config 
[PRODUCT] [PRODUCT_TYPE ] [Camera]   
[PRODUCT] [PRODUCT_MODEL] [WYZEDB3]   
[PRODUCT] [BOARD_TEST   ] [000F]   
[PRODUCT] [FACTORY_TEST ] [0000]   
[PRODUCT] [HARDWARE_VER ] [P00A0000]   
[PRODUCT] [KEY          ] []   
[PRODUCT] [KEY_CHECKSUM ] [12345678]   
[PRODUCT] [CONFIG_INFO  ] []   
[PRODUCT] [macAddr      ] []
[02/22/2021 10:49:02]<Reading> /system/bin/app.ver 
[PRODUCT] [appver       ] [4.25.0.222]   
[02/22/2021 10:49:02]<Reading> /etc/config/.user_config 
[USERSET] [indicator    ] [1]   
[USERSET] [nightVision  ] [3]   
[USERSET] [bitRate      ] [180]   
[USERSET] [res          ] [1]   
[USERSET] [fps          ] [20]   
[USERSET] [horSwitch    ] [1]   
[USERSET] [verSwitch    ] [1]   
[USERSET] [osdSwitch    ] [1]   
[USERSET] [logSd        ] [1]   
[USERSET] [logUdisk     ] [1]   
[USERSET] [telntSwitch  ] [2]   
[USERSET] [recordType   ] [1]   
[USERSET] [MASwitch     ] [2]   
[USERSET] [MALevel      ] [10]   
[USERSET] [AASwitch     ] [2]   
[USERSET] [AALevel      ] [5]   
[USERSET] [SASwitch     ] [2]   
[USERSET] [CASwitch     ] [2]   
[USERSET] [TBStart      ] [0]   
[USERSET] [TBDuration   ] [1440]   
[USERSET] [drawBoxSwitch] [1]   
[USERSET] [timezone     ] [-5]   
[USERSET] [MMALevel     ] [150]   
[USERSET] [AMALevel     ] [100]   
[USERSET] [MAT          ] [0]   
[USERSET] [AST          ] [1]   
[USERSET] [motionTrack  ] [2]   
[USERSET] [motionCruisin] [2]   
[USERSET] [AASX         ] [25]   
[USERSET] [AASY         ] [25]   
[USERSET] [AALX         ] [50]   
[USERSET] [AALY         ] [50]   
[USERSET] [motionback   ] [1]   
[USERSET] [slide_x      ] [0]   
[USERSET] [slide_y      ] [0]   
[USERSET] [preset_x     ] [0]   
[USERSET] [preset_y     ] [0]   
[USERSET] [ubootflag    ] [0]   
[USERSET] [exposureModel] [0]   
[USERSET] [limit_max_x  ] [0]   
[USERSET] [limit_min_x  ] [0]   
[USERSET] [limit_max_y  ] [0]   
[USERSET] [limit_min_y  ] [0]   
[USERSET] [motor_limit  ] [0]   
[USERSET] [resoluselect ] [1]   
[USERSET] [low_bitrate  ] [30]   
[USERSET] [watermark_flag] [0]   
[USERSET] [speaker_vol  ] [50]   
[USERSET] [pir_alaram   ] [0]   
[USERSET] [alarmDate    ] [255]   
[USERSET] [alarmInterval] [180]   
[USERSET] [storage_device] [1]   
[USERSET] [white_switch ] [2]   
[USERSET] [white_brightness] [100]   
[USERSET] [white_face_switch] [1]   
[USERSET] [faceSwtich   ] [1]   
[USERSET] [osdLogoSwitch] [2]   
[USERSET] [firstBindFlag] [0]   
[USERSET] [chime_voice  ] [6]   
[USERSET] [chime_music  ] [1]   
[USERSET] [chime_times  ] [1]   
[USERSET] [chime_bucket_start] [960]   
[USERSET] [chime_bucket_duration] [1440]   
[USERNET] [ssid         ] []   
[USERNET] [password     ] [*]   
[USERNET] [wifiType     ] [5]   
[USERNET] [p2pid        ] []   
[USERNET] [r_encr       ] []   
[USERNET] [sub1gMac     ] []   
----------------------------------------
    WELCOME TO LOCALSDK      ('_)') 
----------------------------------------
    platform: ingenict31 
    version : SDK20190612d 
    update  : Jan 25 2021 (11:29:42) 
----------------------------------------
[SDK-GPIO]dbg: Pin(39)  Lvl(0)  Dir(out)
[SDK-GPIO]dbg: Pin(38)  Lvl(0)  Dir(out)
[SDK-GPIO]dbg: Pin(53)  Lvl(0)  Dir(out)
[SDK-GPIO]dbg: Pin(52)  Lvl(0)  Dir(out)
[SDK-GPIO]dbg: Pin(6)  Lvl(1)  Dir(in)
[SDK-GPIO]dbg: Pin(7)  Lvl(1)  Dir(in)
[SDK-GPIO]dbg: Pin(61)  Lvl(1)  Dir(out)
[SDK-THREAD]dbg: Platform Thread Create OK!  ('_)')
[SDK-THREAD]dbg: Platform Thread Create OK!  ('_)')
[localsdk_get_version][5]
[02/22/2021 10:49:03][init_local_sdk] localsdk version:5
[02/22/2021 10:49:03][init_local_sdk] localsdk version:5
[SDK-THREAD]dbg: Platform Thread Start...
##### sensor type: sc4236 #####
insmod: can't insert '/system/bin/sensor_sc4236_t31.ko': File exists
err: IMP_ISP_AddSensor() fail
err: sample_system_init() failed
[02/22/2021 10:49:03][init_sdk_video] local_sdk_video_init(fps20) fail...
[02/22/2021 10:49:03][init_local_sdk] video init fail...
[02/22/2021 10:49:03]init_all[1921]:init sdk```

gtxaspec, 2021-03-01:
looks like icamera doesn't save logs to /tmp, it saves to a unix socket now?

deftdawg, 2021-05-14:
Wish there were some compatiable prebuilt static mips32(?) arch binaries for curl, lsof, strings, ldd to poke around a bit more...   so far I can 'load' files on by using the doorbell's busybox `wget` against port 8000 of a host running `python -m SimpleHTTPServer` (wget doesn't support SSL so getting things directly from the internet is difficult).

Here's some output from `logcat` in case it's of interest to @gtxaspec 
```
[root@WyzeCam-4CBC:~]# logcat
I/OSD     (  150): IMP_OSD_SetPoolSize:819200
D/IMP-ISP (  150): ~~~~~~ IMP_ISP_Open[280] ~~~~~~~
I/IMP-ISP (  150): IMP_ISP_AddSensor,366: paddr = 0x0, size = 0x6c6d00
I/Alloc Manager(  150): MEM Alloc Method is kmalloc
D/KMEM Method(  150): CMD Line Rmem Size:48234496, Addr:0x05200000
D/KMEM Method(  150): alloc->mem_alloc.method = kmalloc
D/KMEM Method(  150):                   alloc->mem_alloc.vaddr = 0x66608000
D/KMEM Method(  150):                   alloc->mem_alloc.paddr = 0x05200000
D/KMEM Method(  150):                   alloc->mem_alloc.length = 48234496
I/Alloc Manager(  150): MEM Manager Method is continuous
D/System  (  150): IMP_System_Init SDK Version:1.1.0-d842a44-Fri May 15 11:05:48 2020 +0800, built: May 19 2020 14:11:25
D/System  (  150): system_init()
D/System  (  150): Calling DSystem
D/System  (  150): Calling FrameSource
D/System  (  150): [ignored]read /proc/cpuinfo ret is NULL
D/System  (  150): Calling IVS
D/System  (  150): Calling OSD
D/System  (  150): Calling Encoder
D/System  (  150): Calling FB
I/Encoder (  150): encChn=0,srcFrameCnt=3,srcFrameSize=3359232
I/Encoder (  150): encChn=0,srcStreamCnt=2,enc_chn->stream_frame_size=1058048
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=0 success
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=1 success
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=2 success
D/System  (  150): system_bind(): bind DST-OSD-0(4.0.0) to SRC-Framesource-0(0.0.0)
D/System  (  150): system_bind(): bind DST-Encoder-0(1.0.0) to SRC-OSD-0(4.0.0)
I/Framesource(  150): [chn0]: width = 1728 height = 1296
E/VBM     (  150): VBMCreatePool()-0: w=1728 h=1296 f=842094158
E/VBM     (  150): VBMCreatePool()-0: pool->config.fmt.fmt.pix.sizeimage=3359232 sizeimage=3359232
E/VBM     (  150): VBMCreatePool()-0: sizeimage=3359232
I/VBM     (  150): PoolId:0, frame=0xb415b8, frame->priv=0xb415e0, frame[0].virAddr=67709100, frame[0].phyAddr=6301100
I/VBM     (  150): PoolId:0, frame=0xb419a0, frame->priv=0xb419c8, frame[1].virAddr=67a3d300, frame[1].phyAddr=6635300
I/VBM     (  150): PoolId:0, frame=0xb41d88, frame->priv=0xb41db0, frame[2].virAddr=67d71500, frame[2].phyAddr=6969500
I/Encoder (  150): encChn=1,srcFrameCnt=3,srcFrameSize=460800
I/Encoder (  150): encChn=1,srcStreamCnt=2,enc_chn->stream_frame_size=159232
W/Encoder (  150): Jpeg channel will not share buff
I/Encoder (  150): encChn=7,srcFrameCnt=2,srcFrameSize=460800
I/Encoder (  150): encChn=7,srcStreamCnt=1,enc_chn->stream_frame_size=159232
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=3 success
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=4 success
I/OSD     (  150): IMP_OSD_CreateRgn(1138) create handle=5 success
D/System  (  150): system_bind(): bind DST-IVS-0(3.0.0) to SRC-Framesource-1(0.1.0)
D/System  (  150): system_bind(): bind DST-OSD-1(4.1.0) to SRC-IVS-0(3.0.0)
D/System  (  150): system_bind(): bind DST-Encoder-1(1.1.0) to SRC-OSD-1(4.1.0)
I/Framesource(  150): [chn1]: width = 640 height = 480
E/VBM     (  150): VBMCreatePool()-1: w=640 h=480 f=842094158
E/VBM     (  150): VBMCreatePool()-1: pool->config.fmt.fmt.pix.sizeimage=460800 sizeimage=460800
E/VBM     (  150): VBMCreatePool()-1: sizeimage=460800
I/VBM     (  150): PoolId:1, frame=0xba46a0, frame->priv=0xba46c8, frame[0].virAddr=68248400, frame[0].phyAddr=6e40400
I/VBM     (  150): PoolId:1, frame=0xba4a88, frame->priv=0xba4ab0, frame[1].virAddr=682b8c00, frame[1].phyAddr=6eb0c00
I/VBM     (  150): PoolId:1, frame=0xba4e70, frame->priv=0xba4e98, frame[2].virAddr=68329400, frame[2].phyAddr=6f21400
I/ai      (  150): AI Enable: 1
I/Encoder (  150): framePriv->i_fps_num=20, framePriv->i_fps_den=1
D/Encoder (  150): enc_chn->index=0, gopAttr->uGopCtrlMode=2, gopAttr->uGopLength=40, gopAttr->uNumB=0, gopAttr->uMaxSameSenceCnt=1, gopAttr->bEnableLT=0, gopAttr->uFreqLT=0, gopAttr->bLTRC=0
D/Encoder (  150): enc_chn->index=0, enc_chn->chnFpsMask=1, enc_chn->inFrmRate.frmRateNum=20, enc_chn->inFrmRate.frmRateDen=1, enc_chn->setFrmRate.frmRateNum=20, enc_chn->setFrmRate.frmRateDen=1, rcAttr->outFrmRate.frmRateNum=20, rcAttr->outFrmRate.frmRateDen=1
I/Encoder (  150): framePriv->i_fps_num=20, framePriv->i_fps_den=1
D/Encoder (  150): enc_chn->index=1, gopAttr->uGopCtrlMode=2, gopAttr->uGopLength=40, gopAttr->uNumB=0, gopAttr->uMaxSameSenceCnt=1, gopAttr->bEnableLT=0, gopAttr->uFreqLT=0, gopAttr->bLTRC=0
D/Encoder (  150): enc_chn->index=1, enc_chn->chnFpsMask=1, enc_chn->inFrmRate.frmRateNum=20, enc_chn->inFrmRate.frmRateDen=1, enc_chn->setFrmRate.frmRateNum=20, enc_chn->setFrmRate.frmRateDen=1, rcAttr->outFrmRate.frmRateNum=20, rcAttr->outFrmRate.frmRateDen=1
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=1
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=1
I/ai      (  150): AI Enable Chn: 1-0
I/ai      (  150): EXIT AI Enable Chn: 1-0
I/ai      (  150): AI Set Gain: 27
I/ai      (  150): AI Set Vol: 60
I/ai      (  150): AI NS ENABLE: mode = 3
[... 26 lines cut ...]
I/ao      (  150): AO Set Vol: 60
[... cut ...]
I/ai      (  150): AI_EnableAec:AiDevId = 1,aiChn = 0,AoDevId = 0,aoChn = 0
I/ai      (  150): AI AEC ENABLE
```
Voice call end
```
I/ai      (  150): AI_DisableAec:AiDevId = 1,aiChn = 0
I/ai      (  150): AI AEC DISABLE
````

Turning on night vision
```
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=0
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=0
```
Turning off night vision
```
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=1
D/Encoder (  150): do_day_night_change(593):enc_chn->inStat.is_day=1
```

geeknik, 2021-05-14:
> Wish there were some compatiable prebuilt static mips32(?) arch binaries for curl, lsof, strings, ldd to poke around a bit more... 

You can find all sorts of pre-built binaries for mips and other arch here: https://downloads.openwrt.org/releases/packages-21.02/. The ipk files are just gzip archives. 

For example:
https://downloads.openwrt.org/releases/packages-18.06/mips_mips32/base/curl_7.60.0-4_mips_mips32.ipk
https://downloads.openwrt.org/releases/packages-18.06/mips_mips32/packages/lsof_4.89-2_mips_mips32.ipk

deftdawg, 2021-05-14:
Thx @geeknik, I had another go with getting curl to run, no love... It seems like the binary I grab is being treated like a shell script...   I'm not that well versed in what they've done to make it like that, I'd probably have look at the first phase of boot up before they call switch_root to the running config.

From workstation
```sh
curl -L https://downloads.openwrt.org/releases/packages-21.02/mips_mips32/packages/curl_7.76.0-1_mips_mips32.ipk | tar -Ozxf - ./data.tar.gz | tar -zx --strip-components 3
python -m SimpleHTTPServer
```

On the doorbell
```sh
export WEBSERVER=192.168.1.77 # change IP to workstation's IP
wget http://${WEBSERVER}:8000/curl
# Connecting to 192.168.1.77:8000 (192.168.1.77:8000)
# curl                 100% |********************************************************************************************|   # 129k  0:00:00 ETA
./curl # -sh: ./curl: Permission denied  #expected this
chmod +x ./curl 
./curl # ./curl: line 1: syntax error: unexpected word (expecting ")")
sh ./curl  # ./curl: line 1: syntax error: unexpected word (expecting ")")
```

I'd also love to see RTSP running on these cameras also...  since I haven't seen any reference to this project, let me link it here:  
- [RTSP server + Ingenic T20 SDK](https://github.com/geekman/t20-rtspd)

These doorbells are based on the Ingenic SDK, so maybe it can be of help.

geeknik, 2021-05-14:
> Thx @geeknik, I had another go with getting curl to run, no love... It seems like the binary I grab is being treated like a shell script...

Well shucks, I thought I was being helpful here. I guess you can try this guy's repo, he seems to have built quite the collection of mips binaries: https://github.com/darkerego/mips-binaries. I"m all out of ideas after that. Good luck!

attackvictim, 2021-11-05:
It looks like some of you might have what I need.   I have the doorbell, 4 of them now.  Not because I love them but because they keep failing.   I get a QR is incorrect error.  I took the first one apart and used UART to see logs and found that a local cert is failing and not allowing it to connect to the API.  I don’t know the user/pass for this device so i can’t cat the cert to see what’s wrong with it.   Has anyone reversed the FW or know the creds for these devices.   I’m tired of dealing with support with this issue.

[1 more reply]