HclX/WyzeHacks #70: Custom script does not execute?

famewolf, 2021-01-10:
I have config.inc set to:

# Uncomment and update this to run a custom script after the NFS mount finishes.
# The script must already exists and executable. You can always use "/media/mmc"
# to refer the camera folder on the NFS share. Or use "/mnt" to refer the root
# of NFS share. The value of the variable can only contain script file name, no
# command line arguments. The script will be executed in background.
export CUSTOM_SCRIPT='/mnt/pingchk'

The script /mnt/pingchk has:
[root@WyzeCam-A88A:~]# cat /mnt/pingchk
#!/bin/sh
while true
do
ping -c 4 -q 192.168.3.5
if [ $? -ne 0 ]; then
#echo "Rebooting...."
reboot     
else
sleep 5m
fi
done

I  did remote install again to all 4 camera and when I checked after rebooting a camera ps | grep ping did not find the script. Even though I don't have "auto update" enabled shouldn't reruning remote_install copy the current one over?

HclX, 2021-01-12:
Hmm, you need to have the script available at the expected location with correct permissions. Try telnet onto the device and verify the script is there and runnable

famewolf, 2021-01-12:
The script is in /mnt and chmod 755.  I'm able to telnet in and run it manually without issue.  I'm having issue with the nfs server dropping or wifi dropping and the camera's either rebooting to soon or just continuing to write to limbo so I'm trying to ping the nfs server as opposed to the router.

HclX, 2021-01-18:
if you have log sync enabled, you can search "Starting custom script" to see if it gets executed. This feature is working for me all the time so you will have to debug your way out.

closed 2021-02-21 by HclX