HclX/WyzeHacks #71: How do I change the NFS server settings via telnet?

bryantech, 2021-01-11:
It worked for about 18 hours to my QNAP box. I want it to goto my unraid box much more space. I can't get it to take new settings.

HclX, 2021-01-12:
I suspect this may have something to do with the auto config update: when the system start it makes a copy of the config file to nfs share, upon reboot it checks if the copy in nfs share differs to the one on device if yes, it assumes changes are made to the nfs share copy so it copies back to device. Apparently this logic is not right when you make changes on device using telnet.

I will come up with a fix for this.

bryantech, 2021-01-12:
Thank you. I have 7 cameras and I would love to get all of the feeds saved
to my unraid box.

bryantech, 2021-01-18:
Any updates on this issue?

HclX, 2021-01-19:
The latest change is in branch dev_v3 (release 0.5.01) fixes the issue. With this version, you can use either telnet or AUTO_CONFIG feature to update your configurations. Please check the comments in config.inc.TEMPLATE file on how AUTO_CONFIG works.

bryantech, 2021-01-19:
I can't files for branch dev_v3 (release 0.5.01).

bryantech, 2021-01-20:
I am testing on a V2 Cam. I have tried the SD install and it tells me in installed. No telnet access and the camera's yellow light is solid and have to factory reset put an old firmware on it and then I have to resetup in my account. Firmwares 4.9.5.115 and 4.9.6.218.

HclX, 2021-01-21:
To be clear, you are not running v3_init.sh, are you? v3_init.sh is designed specifically for v3 cameras, you only need remote_install.sh for v2 cameras.  Also, 4.9.5.115 is considered a very old build, you should try 4.9.6.218 with remote_install.sh. If you have an SD card inserted while installing, it will generate a install.log which can be helpful debugging when went wrong.

bryantech, 2021-01-21:
2 cams won't upgrade to 4.9.6.218  but I upgraded to 4.9.6.199 on 2 cams. SD install and ./remote_install.sh sees the cams and it makes me think it is installing it. I get a different response and an error on a cam on my that I know is offline.  Still no telnet of SD card access for the NFS share. Not installing V3 files.

closed 2021-02-21 by HclX