HclX/WyzeHacks #72: Feature Request: Delete recordings older than X days

mpatton125, 2021-01-19:
As per the title, rather than archiving the old footage - would it be possible to add a "config.inc" option to delete recordings older than X configured days?

I have 2 cams and they are currently taking up around 4TB of storage.  Yes, I can delete them manually or run a cronjob etc - but it would be great for this feature to be implemented in the hack.

MelonSmasher, 2021-02-26:
I'm running this script on a cronjob from a Ubuntu install on a Raspberry Pi. I'm not sure if this will run on a camera but it can defiantly be run from macOS or a Linux machine that has the same NFS share mounted. If someone is up to it, I'm sure it could be implimented within the camera, I'm just not familiar with how to port it into the hack. The script lets you chose the amount of hours you'd like to keep recordings around and then prunes empty directories from the recordings directory.

[MelonSmasher/wyze_rec_cleaner](https://github.com/MelonSmasher/wyze_rec_cleaner)

HclX, 2021-03-14:
the camera doesn't have much resource, running video recording is already taking most of the CPUs so I'd suggest some script on NFS server side is a better solution.

closed 2021-03-14 by HclX