HclX/WyzeHacks #73: "SD card not available" with firmware 4.9.6.235

unomar, 2021-01-24:
My v2 cams updated to 4.9.6.235 and suddenly no longer recognize the NFS mount as a SD card.  NFS is still mounted on the device, just not recognized as SD card.

Pan cam with 4.29.4.41 still works ok.

HclX, 2021-01-26:
That sounds like a beta firmware?  In general I don't test too much with beta firmware, however, thanks for reporting me this.

unomar, 2021-01-26:
Yes, I should have specified.  I had actually forgotten that I was running the beta app.  Much like I just realized that my Pan cam firmware is still the old RTSP version.

twig123, 2021-02-25:
This same thing happens with the newer v2 beta firmware as well (4.9.6.239).

The strange thing though, is that NFS partially works...

- alarm
  - Not working
- archive
  - Created properly, but not working as there are no new recordings
- record
  - Not working
- time_lapse
  - Not working
- wyzehacks
  - Created properly
- .max_############
  - Created properly
- ifconfig.txt
  - Created properly
- install.log
  - Created properly

I'm assuming the root cause of the not working folders is the "SD Card Emulation" issue?

I rolled the v2 cam back to the non-beta 4.9.6.199, and confirmed that SD Card Emulation and NFS are working properly in that firmware (I haven't tested the newer public builds though)

HclX, 2021-03-14:
beta software is usually not supported, but i believe the latest release should work for future beta firmwares if there is no big changes. I'm closing this now.

closed 2021-03-14 by HclX