HclX/WyzeHacks #74: WyzeCamV3: /tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory

hyukishi, 2021-01-27:
I'm trying to install WyzeHacks on my V3. The install.log contains the following:
`+ echo WyzeHack: Starting wyze hack installer...
WyzeHack: Starting wyze hack installer...
+ play_sound /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/begin.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ [ -f /media/mmc/debug/.copyfiles ]
+ echo WyzeHack: Enabling telnetd...
WyzeHack: Enabling telnetd...
+ telnetd
+ set_passwd root::10933:0:99999:7:::
+ /bin/umount /etc
+ rm -rf /tmp/etc
+ cp -r /etc /tmp/
+ echo root::10933:0:99999:7:::
+ /bin/mount -o bind /tmp/etc /etc
+ [ -z 4.36.0.228 ]
+ echo WyzeHack: Current Wyze software version is 4.36.0.228
WyzeHack: Current Wyze software version is 4.36.0.228
+ echo WyzeHack: Installing WyzeHacks version 0.5.01
WyzeHack: Installing WyzeHacks version 0.5.01
+ [ -f /params/wyze_hack.cfg ]
+ [ -f /tmp/Upgrade/config.inc ]
+ echo WyzeHack: Use config file /tmp/Upgrade/config.inc
WyzeHack: Use config file /tmp/Upgrade/config.inc
+ sed s/\r$// /tmp/Upgrade/config.inc
+ [ -f /media/mmc/config.inc ]
+ [ ! -f /configs/wyze_hack.cfg ]
+ [ -z V3 ]
+ echo WyzeHack: Copying wyze hack binary...
WyzeHack: Copying wyze hack binary...
+ cp /tmp/Upgrade/wyze_hack.sh /configs/wyze_hack.sh
+ echo WyzeHack: Hooking up boot script...
WyzeHack: Hooking up boot script...
+ hook_init
+ grep wyze_hack.sh /etc/init.d/rcS
if [ -f /configs/wyze_hack.sh ]; then
    /configs/wyze_hack.sh &
+ return 0
+ play_sound /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 1
+ /tmp/wyze_hack/install/bin/audioplay /tmp/Upgrade/install_snd/finished.wav 50
/tmp/wyze_hack/install/main.sh: line 1: can't create /sys/class/gpio/gpio63/value: nonexistent directory
+ echo 0
+ rm /media/mmc/version.ini.old
+ mv /media/mmc/version.ini /media/mmc/version.ini.old
+ /sbin/reboot
`

To clarify, I did follow the install instructions specific to the V3 in the tools/v3/README.md. I applied the special firmware from the tools/v3/firmware folder and then proceeded to run the v3_init.sh script followed by the remote_install.sh script.

Any help is appreciated.

hyukishi, 2021-01-27:
While it appears that the only functionality that's missing is the ability to play those sound files included in the firmware.bin file, it is worth noting that telnet works after using the v3_init.sh but does not work after running remote_install.sh

HclX, 2021-01-30:
Yes, this is a know issue: the gpio port is not available for v3 camera, and the sound player does not work either. However, that shouldn't affect the functionality.

hyukishi, 2021-01-31:
The functionality has been impacted, perhaps not by the sound player or lack of gpio related to the speaker, but I cannot telnet into the camera and do not see any data on my NFS v3 share after completing the remote install script.

HclX, 2021-03-14:
other than the error message and sound player issue, things are working perfectly on the v3 cameras i have. I'm closing this one since there is not much i can do with it.

closed 2021-03-14 by HclX