HclX/WyzeHacks #75: Find a way to allow multiple resources access to /dev/videoX

gtxaspec, 2021-02-03:
I'm not sure if this is the right place for this, but the community here seems to have some good ideas, perhaps we can get something moving on this front.   Allow multiple resources access to /dev/videoX to enable rtsp/hls/mjpeg/etc servers.

/dev/videoX is not accessible on the Wyze Cameras currently, `can't open '/dev/video0': Device or resource busy`, as the node is in use by the iCamera process.

in the RTSP firmware, the iCamera process looks like it will bind to two unix sockets, /tmp/mediaServer/live.264 and /tmp/mediaServer/live.g711, and the rtsp server reads from those two locations.

These sockets are absent on the standard firmware.

A possible option, a v4l2 loopback node using a special kernel module.  ref https://github.com/umlaeute/v4l2loopback/

I don't have a proper computer at the moment where I could setup a build system with the right toolchains to compile this to try it, so it may be worth a shot if someone has the capability to do this in the meantime.

endertable, 2021-02-07:
This would be awesome.  I would love to have something other than RTSP which is slow and unreliable.

endertable, 2021-03-11:
I think I have a v4l2loopback.ko that creates new /dev/videoX, but where do we go from here.  These work by being fed video from something else, no?  What would we use to feed it video as iCamera locks up video0 - video2.  I am hoping we can get an mjpeg stream or anything more "live".  Any ideas?

gtxaspec, 2021-03-12:
the module should be loaded before iCamera starts.  did you manage compile this module?

endertable, 2021-03-12:
> 
> 
> the module should be loaded before iCamera starts. did you manage compile this module?

OK, I'll try that. does that mean iCamera will automatically feed it?  iCamera automatically creates and uses video0-video2, so do I create those before iCamera or do I force video3, video4, etc.  Is there a special depmod or modprobe command I need to use?

HclX, 2021-03-14:
i don't think it falls into the scope of this hack.

endertable, 2021-03-20:
@gtxaspec Hi, sounds like the owner wants to close this Issue out pretty soon.  Was wondering if you had any ideas on how we can use the v4l2loopback.ko created devices to work with the stock iCamera or if it is even possible.  The bad news for V3 is, they aren't using the /dev/video devices, but maybe for V2 we can still do something.  Any thoughts?

gtxaspec, 2021-05-17:
solution in progress by https://github.com/FiveLeavesLeft/WyzeCameraLiveStream

closed 2021-05-17 by gtxaspec