HclX/WyzeHacks #76: Which instructions to follow for WyzeCam v3?

SomebodySysop, 2021-02-03:
I have WyzeCam V3 camera with firmware 4.36.0.228 installed (installed using Wyze update).

The instructions here are for using remote_install.sh: https://github.com/HclX/WyzeHacks

The instructions here require installing firmware on sd card and running init_v3.sh: https://github.com/HclX/WyzeHacks/blob/master/tools/v3/README.md

Which instructions do I follow to install WyzeHacks?

SomebodySysop, 2021-02-04:
OK, then let me ask it this way:

These instructions:

"First, use the SD card recovery method to install firmware 4.36.0.228 (you can find the image in tools/v3/firmwares), just like what you'd do with v2 cameras except that you need to name the firwmare demo_wcv3.bin on SD card."

If I already have the Wyze 4.36.0.228 firmware installed, do I still need to do this step?

If so, what is the "SD card recovery method"? 

Thanks for any assistance!

HclX, 2021-02-04:
"SD card recovery method" is the wyze official device recovery method, used to recovery a firmware if the device is bricked. It's used here to bring the camera to a known version of firmware before we start doing anything.

closed 2021-02-04 by HclX

SomebodySysop, 2021-02-04:
If I already have the Wyze 4.36.0.228 firmware installed, do I still need to do this step?

HclX, 2021-02-04:
In that case you can skip that step.