HclX/WyzeHacks #77: unable to use v3_init, python error.

gtxaspec, 2021-02-05:
`Exception happened during processing of request from ('10.25.1.182', 55103)
Traceback (most recent call last):
  File "/usr/lib/python3.7/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.7/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.7/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.7/http/server.py", line 646, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.7/socketserver.py", line 720, in __init__
    self.handle()
  File "/usr/lib/python3.7/http/server.py", line 426, in handle
    self.handle_one_request()
  File "/usr/lib/python3.7/http/server.py", line 414, in handle_one_request
    method()
  File "./wyze_updater.py", line 188, in do_GET
    self.wfile.write(firmware_data)
  File "/usr/lib/python3.7/socketserver.py", line 799, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe
`

using python 3.7

gtxaspec, 2021-02-05:
issue went away, yay!

closed 2021-02-05 by gtxaspec

stephen-yeung, 2021-05-20:
I am getting exact same error. how did you manage to resolve it?

gtxaspec, 2021-05-20:
fix the python environment, probably you have an older version?

stephen-yeung, 2021-05-20:
which version did you use? i am using 3.7.0