HclX/WyzeHacks #79: Installation failed using remote install. (WyzeCam V2

vabajaa, 2021-02-06:
It keeps saying installation failed after using the remote install guide on my Wyze Cam V2.
Also, can I install it with NFS disabled at first? Or does it have to be enabled and set up to install correctly, I've gotten fairly lost trying to understand things.

tasteItheIvalue, 2021-02-06:
More info needed... Current firmware version? Are you using the latest package from here? What kind of machine are you doing the remote install from? (Dev stated Linux or maybe Mac, but Windows using WSL doesn't work.)

vabajaa, 2021-02-06:
The V2 Camera is running version 4.9.6.218 I am running the latest version of WyzeHacks and I am using a raspberry pi running Ubuntu 20.10 for installing remotely.

HclX, 2021-02-08:
try have an sdcard on the camera while you are doing remote install, there will be install.log created on the SD card showing what's wrong. If this is the first time you install it, i'd check if you have config.inc file correctly placed.

vabajaa, 2021-02-11:
Okay so I've tried doing that and now the camera doesn't even say anything when doing remoteinstall.sh

Instead, it outputs this into the console before moving onto the next camera

INFO:root:Serving firmware file './firmware.bin' as 'http://192.168.1.13:11808/firmware.bin', md5=23b2dafec72c917d612b71db77007662
INFO:root:Checking device, mac=7C78B2126C98

HclX, 2021-02-21:
not sure what's going on with yours, it seems you might have something in your network preventing the camera from getting the firmware. check firewall or router settings?

revenant-81, 2021-02-22:
Not sure if this will help, but from personal experience I had to disable "Block LAN to WLAN multicast and broadcast data" (Unifi USG router) as it was causing issues with the Ubuntu box sending the file over to the Wyzecams with it on. Also ensure the Linux box is on the same subnet as the cameras you wish to push to.

vabajaa, 2021-02-22:
I'll try that and see if it gives a log right now.

vabajaa, 2021-02-22:
Currently trying to flash the firmware from wyze.com and reset & see if that helps.

vabajaa, 2021-02-22:
Weird, the camera now refuses to revert versions from the app & won't take loading new versions via SDcard

HclX, 2021-03-14:
it works perfectly on all my cameras, and others. I'm going to close this one since there is not much I can do with it.

closed 2021-03-14 by HclX