HclX/WyzeHacks #80: Cams do not recognize nfs has dropped.

famewolf, 2021-02-07:
When my raspberry pi that provides the nfs share reboots the cam's do not realize this and continue in a hung state.  If you connect to them in the app and try to view playback it will tell you there is no sdcard.  I realize there is code that is supposed to handle a stale nfs link but I'm not sure if it's working right and if it is I'm wondering if the cams are rebooting before the pi has time to come back up itself to restore the nfs mount.    I ended up adding some lines right after the nfs mount on the PI which hopefully will force all the cams to reboot.  Repeated as necessary for each cam....

(sleep 3;echo root;sleep 3;echo xxx;sleep 3;echo reboot;sleep 3;) | telnet 192.168.3.20

HclX, 2021-02-09:
interesting, I made a fix in "bugfixes" branch, can you try it out?

HclX, 2021-02-09:
the fix will wait for 20 seconds and then reboot the camera.

famewolf, 2021-02-09:
Sure...timing is good since came do a nightly reboot in about an hour.

HclX, 2021-02-09:
Also, remember you can now use the auto update feature to install new
build:)

famewolf, 2021-02-09:
I just usually push the install....but there does not seem to be a standard zip available for download in bugfixes branch.  Given it's 2am here maybe I'm missing something?

HclX, 2021-02-09:
It's not your problem but mine, try refresh the page:)

famewolf, 2021-02-09:
Ok I tried it..rebooted the pi and everything worked like it should....then I rembered I need to edit the nfsmount service setup to mount nfs because it reboots the cams using the script example I showed.   I'm a little to tired to be editing files at the moment so I'll tackle this later in the day after some sleep and report back.  I think your solution will work fine for most cases.

HclX, 2021-02-09:
No worries, time to sleep...

famewolf, 2021-02-10:
Just tested this again with my reboot cam lines commented out and they did fine....they rebooted and reconnected to the pi nfs share...I verified by going to a cam in the wyze app and clicking to view playback because when in a hung state if you telnet in and df it wlll show it's mounted...if you do "touch test"  that test won't show up on pi....and the app will say no microsd (even though it actually has one...would still love to mount it somewhere else and copy the 1 minute files as a backup.)

HclX, 2021-02-10:
So the fix works?

famewolf, 2021-02-10:
Seemed to work fine in my quicky test.   If you want I can monitor it over the next few days and do some random reboots of the nfs server......you could simulate the same thing by disabling the nfs service on your pc for around 20 seconds then turn it back on.

famewolf, 2021-02-10:
Or wait a full 60+ seconds and see if the camera gets an error when it tries to write to the nfs share.  I would have thought that returned an error vs a stale nfs handle.

HclX, 2021-02-11:
I was trying this with my NAS and there are still issues:
1. I simply restart my NAS and the camera was able to detect the NFS share is offline, and reboot correctly.
2. However, after reboot, mounting NFS always returns stale file handle.

I was actually hit by this as all the cameras are trying to mount the NFS share forever after I rebooted my NAS two days ago. I didn't noticed that until today I noticed all the cameras don't have SD card mounted.

I googled around and this seems to be a problem on the NFS server side, not the camera side. The "un-export/re-export" trick from this page seems to be fixing the issue:
https://unix.stackexchange.com/questions/433051/mount-nfs-stale-file-handle-error-cannot-umount

famewolf, 2021-02-11:
I've added those lines to my /etc/init.d/nfsmount service script.  Perhaps that should be mentioned in the documentation to be done when the nfs is mounted.

#! /bin/sh
# /etc/init.d/nfsmount

### BEGIN INIT INFO
# Provides:          server
# Required-Start:    $remote_fs $syslog
# Required-Stop:     $remote_fs $syslog
# Default-Start:     2 3 4 5
# Default-Stop:      0 1 6
# Short-Description: Simple script to start a program at boot
# Description:       ..
### END INIT INFO

# If you want a command to always run, put it here
mount /mnt/nfsshare
exportfs -ua
# cat /proc/fs/nfs/exports
sleep 2s
exportfs -a
#(sleep 3;echo root;sleep 3;echo xxx;sleep 3;echo reboot;sleep 3;) | telnet 192.168.3.20
#(sleep 3;echo root;sleep 3;echo xxx;sleep 3;echo reboot;sleep 3;) | telnet 192.168.3.21
#(sleep 3;echo root;sleep 3;echo xxx;sleep 3;echo reboot;sleep 3;) | telnet 192.168.3.193
#(sleep 3;echo root;sleep 3;echo xxx;sleep 3;echo reboot;sleep 3;) | telnet 192.168.3.227

famewolf, 2021-02-12:
A thought occured to me....what happens if the router between the camera and the nfs server reboots?

revenant-81, 2021-02-15:
Running the latest bugfix release, I noticed the NFS connected camera rebooting more than usual. Probably due to losing the connection to the NFS server. Think it would be possible to set the NFS timeout value as variable instead?

HclX, 2021-02-16:
@revenant-81  Sure, i will see if i can come up with a quick fix on this. Meanwhile, you may want to try the "ping hack" (check config template) which solves the disconnect issue for my setup.

famewolf, 2021-02-16:
I've been without internet for about the last 3 days thanks to a damaged cell tower.   I called in after 24 hours to find out what was going on and they said no one had reported that issue but there was a definate problem.   Don't they have all this fancy equipment to tell them this before cranky customers call in?   In any case I noticed just now reviewing cam history they do not appear to have written to the nfs share even though the router, cams and nfs server were all up during this time period even if  no internet is available.   This is a MAJOR security flaw if confirmed.  All someone would need to do is bring one of those wifi nullifiers they can build for around $20 in parts and  instructins from the internet to cause my entire security system to go down.   PLEASE for the love of whatever deity  you pick give us the sdcard mounted to an alternative mount and the 1 minute file copied over...I don't even care if it's the 1 minute OLD video file as long as we have a way to get footage when these things happen because they are happening ALOT.

HclX, 2021-02-22:
@famewolf was there any logs on the NFS share? Internet down shouldn't affect NFS functionalities.

famewolf, 2021-02-22:
I recently had to completely redo my router so any logs are now history along with alot of customized info. If I catch it again I'll post more. Quite possible it was a router issue since it was giving seg faults running apps in ssh session.