HclX/WyzeHacks #83: telnet is disabled if NFS is not accessible

gtxaspec, 2021-02-10:
telnet is disabled if NFS is not accessible...

re did some configurations today and forgot to allow access to the VPN network...so the NFS share was not accessible from the camera.  this had the side effect of not enabling telnetd, so i thought it was a camera issue instead of a network issue.  

verified on another camera, if NFS doesn't mount, telnet doesnt work.

gtxaspec, 2021-02-11:
Just noticed there is a new bugfix release, I will try to see if the issue is fixed there!

HclX, 2021-02-11:
no, it's not fixed yet. actually i just found this on my cameras, will come up with a fix in bugfix branch.

HclX, 2021-02-11:
the fix is now in `bugfix` branch, please give it a try.

gtxaspec, 2021-02-16:
confirmed fixed

closed 2021-02-16 by gtxaspec